raspicamplayer/camplayer #2: Issue detecting screen resolution

XIRGO, 2020-06-30:
I'm currently coming across an issue on a raspberry pi 4 model b v1.2 where the screen resolution is not properly being detected.

I have pipng installed which does appear to properly detect the screen resolution. The below was captured from the logs while running the camplayer manually.

My screen resolution being used is 1440x900. Is there an override option available that would allow for the resolution to be set manually through the config.ini file?

```
2020-06-29 20:16:12.711421 - Main - INFO - Starting camplayer version 1.0.0b2
2020-06-29 20:16:12.711551 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2020-06-29 20:16:14.413912 - Main - INFO - ********************** SYSTEM INFO **********************
2020-06-29 20:16:14.414186 - Main - INFO - Camplayer version             = 1.0.0b2
2020-06-29 20:16:14.414267 - Main - INFO - Operating system              = Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
2020-06-29 20:16:14.414372 - Main - INFO - Raspberry Pi SoC              = BCM2835
2020-06-29 20:16:14.414453 - Main - INFO - Raspberry Pi revision         = c03112
2020-06-29 20:16:14.414575 - Main - INFO - Raspberry Pi model name       =
2020-06-29 20:16:14.414672 - Main - INFO - GPU memory allocation         = 256 MB
2020-06-29 20:16:14.414800 - Main - INFO - Python version                = 3.7.3 (default, Dec 20 2019, 18:57:59)  MB
2020-06-29 20:16:14.414884 - Main - INFO - VLC installed                 = True
2020-06-29 20:16:14.414965 - Main - INFO - pipng installed               = True
2020-06-29 20:16:14.415081 - Main - INFO - ffmpeg installed              = True
2020-06-29 20:16:14.415149 - Main - INFO - *********************************************************
2020-06-29 20:16:14.416672 - Main - INFO - advanced setting overwritten for 'hardwarecheck' is '0'
2020-06-29 20:16:14.438469 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2020-06-29 20:16:14.438809 - Main - INFO - Using a virtual screen resolution of '1920x1080'
```

raspicamplayer, 2020-06-30:
Is the following line commented out in '/boot.config.txt' as described in the readme?
`# dtoverlay=vc4-fkms-v3d`

But it is possible to overwrite the detected resolution by adding some advanced settings to your config file:
```
[ADVANCED]
screenwidth = 1440
screenheight = 900
```

XIRGO, 2020-06-30:
> 
> 
> Is the following line commented out in '/boot.config.txt' as described in the readme?
> `# dtoverlay=vc4-fkms-v3d`
> 
> But it is possible to overwrite the detected resolution by adding some advanced settings to your config file:
> 
> ```
> [ADVANCED]
> screenwidth = 1440
> screenheight = 900
> ```

I double checked the /boot/config.txt file just be sure and I discovered that there are actually two instances of the `dtoverlay=vc4-fkms-v3d` setting that are enabled by default. See below

```bash
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem=256
hdmi_enable_4kp60=1
```
nevertheless commenting out these two did not solve the issue with the auto detection feature.

Adding the screenwidth and screenheight parameters under the ADVANCED section did the trick. 

Thanks for the support!

closed 2020-06-30 by XIRGO