raspicamplayer/camplayer #7: NVR Stream NO video Issue

imma83, 2021-02-16:
I have am facing a small issue when I try to use rtsp stream from Hikvision NVR there is no video output below is the url I am using 

channel1.1_url = rtsp://user:User1234@192.168.88.105/ISAPI/Streaming/Channels/402 (main stream)
channel1.2_url = rtsp://user:User1234@192.168.88.105:554/Streaming/Channels/402 (sub stream)
The above two URL there is no video 

But all the other streams directly from the cameras are working , below is are url used for cameras 

[DEVICE3]
channel1_name = optional OSD name
channel1.1_url = rtsp://admin:User1234@192.168.88.101:554/cam/realmonitor?channel=1&subtype=1 (main stream)

channel1.1_url = rtsp://admin:User1234@192.168.88.100:554/Streaming/Channels/2 

can you please kindly help me on this

raspicamplayer, 2021-02-16:
Hi @imma83 
Can you run the following commands (with the correct username and password) and post its output?
`ffprobe -hide_banner rtsp://user:User1234@192.168.88.105/ISAPI/Streaming/Channels/402`
`ffprobe -hide_banner rtsp://user:User1234@192.168.88.105:554/Streaming/Channels/402`

Be careful and mask your credentials in the output ;-)

imma83, 2021-02-16:
> Hi @imma83
> Can you run the following commands (with the correct username and password) and post its output?
> `ffprobe -hide_banner rtsp://user:User1234@192.168.88.105/ISAPI/Streaming/Channels/402` [...]

[
[image]

Please find the attached output

raspicamplayer, 2021-02-19:
The video format looks fine.
Can you try playing the streams directly with omxplayer (the media used by camplayer)?
`omxplayer rtsp://user:User1234@192.168.88.105/ISAPI/Streaming/Channels/402`
`omxplayer rtsp://user:User1234@192.168.88.105:554/Streaming/Channels/402`

Another thing to try is posting the output when you run camplayer in a terminal window:
`sudo systemctl stop camplayer.service` (just in case)
`camplayer`

SaifAqqad, 2021-02-27:
I'm also using a Hikvision nvr but my urls have this format 
`rtsp://username:password@192.168.0.220:554/Streaming/Unicast/channels/101` for the main stream
`rtsp://username:password@192.168.0.220:554/Streaming/Unicast/channels/102` for the sub stream

civita, 2021-03-08:
I faced the video playback issues with specific cameras, which use rtsp over udp. After commenting out the options `--avdict rtsp_transport:tcp` when running `omxplayer` in `windowmanager.py`, it works and starts showing the video. Not sure if OP faced the same issue or not. Would like to have option for each `[DEVICEx]` to choose what protocol rtsp based on.

raspicamplayer, 2021-03-16:
@civita @imma83 I've added a force UDP setting, can you update camplayer and report back?  
`
[DEVICE1]
channel1_name = optional OSD name
**channel1_force_udp = 1**
channel1.1_url = your 1st IP camera stream URL (main stream)
channel1.2_url = your 1st IP camera stream URL (sub stream)
`

imma83, 2021-03-17:
> @civita @imma83 I've added a force UDP setting, can you update camplayer and report back?
> 
> Format:
> [DEVICE1]
> channel1_name = optional OSD name
> channel1_force_udp = 1
> channel1.1_url = your 1st IP camera stream URL (main stream)
> channel1.2_url = your 1st IP camera stream URL (sub stream)

It still doesnt work

raspicamplayer, 2021-03-18:
> The video format looks fine.
> Can you try playing the streams directly with omxplayer (the media used by camplayer)?
> `omxplayer rtsp://user:User1234@192.168.88.105/ISAPI/Streaming/Channels/402` [...]
@imma83

Haglerd, 2021-03-20:
> The video format looks fine.
> Can you try playing the streams directly with omxplayer (the media used by camplayer)?
> `omxplayer rtsp://user:User1234@192.168.88.105/ISAPI/Streaming/Channels/402` [...]

What is supposed to happen when you play it using OMXplayer? I'm getting same issues. When I do that nothing shows up at all on demo or using the omxplayer.

RuhanSA079, 2021-11-08:
@imma83 I know this is an old thread, did you check on another device that the NVR stream is working? Please ensure that the RTSP encryption on the NVR is disabled. (Had to do it on mine)