raspicamplayer/camplayer #8: Does not run on Rpi400

jsrober, 2021-02-21:
pi@raspberrypi:~ $ camplayer --demo
2021-02-21 16:39:56.375710 - Main - INFO - Starting camplayer version 1.0.0b3
2021-02-21 16:39:56.375851 - Main - INFO - Using config file '../examples/demo-config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-02-21 16:39:58.147605 - Main - INFO - ********************** SYSTEM INFO **********************
2021-02-21 16:39:58.147899 - Main - INFO - Camplayer version             = 1.0.0b3
2021-02-21 16:39:58.147977 - Main - INFO - Operating system              = Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
2021-02-21 16:39:58.148061 - Main - INFO - Raspberry Pi SoC              = BCM2711
2021-02-21 16:39:58.148131 - Main - INFO - Raspberry Pi revision         = c03130
2021-02-21 16:39:58.148192 - Main - INFO - Raspberry Pi model name       = 
2021-02-21 16:39:58.148263 - Main - INFO - GPU memory allocation         = 76 MB
2021-02-21 16:39:58.148345 - Main - INFO - Python version                = 3.7.3 (default, Jul 25 2020, 13:03:44)  MB
2021-02-21 16:39:58.148420 - Main - INFO - VLC installed                 = True
2021-02-21 16:39:58.148488 - Main - INFO - pipng installed               = False
2021-02-21 16:39:58.148548 - Main - INFO - ffmpeg installed              = True
2021-02-21 16:39:58.148609 - Main - INFO - *********************************************************
Unsupported hardware with revision c03130 ...

raspicamplayer, 2021-02-22:
Rpi400 is not officially supported, but I suppose I would run fine if you disable the HW check.
Add the following to your config file ('../examples/demo-config.ini' in your case):
```
[ADVANCED]
hardwarecheck = 0
```
Also, your GPU memory allocation is insufficient. See https://github.com/raspicamplayer/camplayer#installation

jsrober, 2021-03-27:
Thank you for your help. I followed the instructions and it's working on a single monitor.

I fund that commenting out dtoverlay=vc4-fkms-v3d caused my second monitor to stop working. 

I'd like to be able to run two monitors with camplayer. I have a 42" TV and a 55" TV both connected to the Raspberry Pi 400.

Is this possible?

John

raspicamplayer, 2021-05-02:
@jsrober Rpi400 is supported if you update/reinstall camplayer software.
For dual display, yes that should be possible but with a performance impact: https://github.com/raspicamplayer/camplayer#dual-display-1920x108060hz

closed 2021-05-02 by raspicamplayer