raspicamplayer/camplayer #10: Enhancement: Touchscreen/mouse interactions instead of keyboard

jschwerdlin, 2021-03-12:
I've got the 7" Pi touchscreen and was hoping to build this into the wall as a persistent display for my cameras, but would love to be able to interact via the touchscreen rather than a keyboard for obvious reasons

Ideal actions:
Double click on small video in grid to make it full screen
Double click on full screened video to go back to grid
Swipe left/right to go to the next/previous page of grid view or go to the next/previous camera in full screen mode

jschwerdlin, 2021-03-13:
FYI - I was able to hack something together by editing inputhandler.py, but it is VERY specialized to the RPi official 7" 800x480 screen with exactly 4 streams being shown.  I only got the double click part working and it can be a bit flakey 

Basically, in your event loop I added listening for BTN_TOUCH and EV_ABS.  Any time I get a new position with ABS_X or ABS_Y I store it off.  

I check for two clicks (BTN_TOUCH) events within 500ms of each other to declare a double click.  If so and if we're already full screen, I inject a "0" keypress events into your queue to go back to the grid.  If we're already in a grid, I inject a 1-4 keypress based on dividing the screen into quadrants and the most recent position reading to full screen the right stream.

It can be a bit flakey on the injection for some reason.  It always detects the double click, but it seems that the event doesn't always cause an action to take place.

So, like I said, very specialized to that screen only (not a general screen and not a mouse) and to a 2x2 setup, but it could definitely be generalized.  And maybe there is a more efficient way to trigger the events than injecting a fake keypress into the queue

raspicamplayer, 2021-03-16:
@jschwerdlin Thanks for sharing, I may implement something similar for V2.

jschwerdlin, 2021-03-17:
FYI - I did fork your code and have committed into my fork so you can see the mods (I didn't do a pull request though).  I fixed the flakey event injection and also was able to add swipe left/right support to change pages.

cybermaus, 2021-07-15:
@raspicamplayer  If you do get round to implementing such. I would suggest a mouse driver. My guess is that would also work for the touch screen, and it would work for normal monitors too, a cheap (wireless) mouse mouse is easier/cheaper/smaller to have nearby each monitor then a keyboard.

I am trying to figure out myself how to do this, but I am a bit of a noob in python. But I did get one fix/PR already, so who knows.

cybermaus, 2021-07-17:
I made a PR with a mouse gesture driver. #48 

Somewhat similar to above, contained in InputHandler and injecting keystrokes. But using relative instead of absolute positioning. It should work for all grids and screen sizes.

Wilksj01, 2022-01-05:
@jschwerdlin I've just used your fork and it works great on a 1600x1200 touchscreen.  @raspicamplayer thanks for making this great software.

jschwerdlin, 2022-01-06:
@Wilksj01 Glad you found it helpful.  I'm guessing you had to make some mods though since the coordinate division is extremely tied to the 800x480 Pi screen, though I think it would be a really simple mod to work on a larger screen

Wilksj01, 2022-01-06:
@jschwerdlin