raspicamplayer/camplayer #11: No video in Demo

stephbairey, 2021-03-17:
Help? I've got a fresh install of Rasbian Lite on a Rpi4, hooked to a 1920x1080 display by HDMI. The display works, shows login etc if I don't have #dtoverlay=vc4-fkms-v3d commented out. I've followed the install, settings file contains only 1 device, set to display 1 cam, rtsp stream verified working. Everything upgraded. No display appears.

Pastebin of configs and results: https://pastebin.com/1qUmTkwy

I stepped back to try the demo again, and am getting many instances of this: 

Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S03_W01' is not responding correctly

Pastebin of demo logging: https://pastebin.com/gzq8cBb9

raspicamplayer, 2021-03-18:
A few things to try:
* change 'max_framebuffers=2' to  'max_framebuffers=1' in config.txt
* change 'hdmi_mode=82' to 'hdmi_mode=16' in config.txt
And then test with demo mode again.

stephbairey, 2021-03-18:
Thank you, it's closer, but not quite there. The display now shows the Pi splash screen on reboot (instead of just black) but it doesn't go away. Running demo gets me more of:

```
2021-03-18 18:48:02.666722 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S03_W01' is not responding correctly, retrying within 250ms
2021-03-18 18:48:02.897368 - Window - ERROR - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S02_W03' is not responding correctly after '6' attemps, give up now
```

raspicamplayer, 2021-03-18:
Hmm, just to be sure, you are using the HDMI0 output?

stephbairey, 2021-03-18:
Good lord, that was it. New case has the HDMI noted to the side and I misinterpreted. Demo is now working! 

Is it expected to show the CLI in sections where there is no video? (see attachment) I realize I can install the pipng, maybe that changes that.

Working now on getting my own video to show and to get it fullscreen. Can I tip for support?
[image]

stephbairey, 2021-03-18:
Apologies if this isn't where to ask - everything seems to be working now, all demo videos run, I've got pipng installed and working, screen is running at 1920x1080. BUT, my one cam won't display. I'm getting the NO LINK screen. I can stream the rtsp on VLC, and pinging the IP form the RPI works. Everything's on the same local network. Cam is hardwired, RPI is over 5G wifi. I do only have one stream, not a substream, so my settings are: 

```
[DEVICE1]
channel1_name = 7-11
channel1.1_url = rtsp://192.168.0.192:554/stream1
 
[SCREEN1]
layout = 1
window1 = device1,channel1

[ADVANCED]
showtime = 15
hardwarecheck = 1
icons = 1
screenwidth = 1920
screenheight = 1080
enablevideoosd = 1
```

Camplayer is saying: https://pastebin.com/3U8rqTr3

It seems it can find the video, from this line: 

```
2021-03-18 21:40:57.724583 - StreamInfo - INFO - stream properties 'rtsp://192.168.0.192:554/stream1', resolution '2304x1296@25', codec 'h264', calculated weight '0', valid url '1', has audio 'True', valid video 'window
2021-03-18 21:40:57.724583 - StreamInfo - INFO --> ed 0 fullscreen 0'
```

Ideas? Thanks!

Haglerd, 2021-03-20:
> Good lord, that was it. New case has the HDMI noted to the side and I misinterpreted. Demo is now working!
>
> Is it expected to show the CLI in sections where there is no video? (see attachment) I [...]

So what did you do to fix it?

raspicamplayer, 2021-03-21:
> Apologies if this isn't where to ask - everything seems to be working now, all demo videos run, I've got pipng installed and working, screen is running at 1920x1080. BUT, my one cam won't display. [...]

Your stream is exceeding the capabilities of the Raspberry Pi hardware '2304x1296@25', codec 'h264'. Max resolution for h264 is 1920x1080. And even less if you display multiple streams simultaneously. You can probably adjust this in your camera.

stephbairey, 2021-03-21:
> So what did you do to fix it?

There are two possible HDMI ports. I was using the wrong one. When I switched the plug, it worked.

marcgrc, 2021-03-28:
[image][image]

marcgrc, 2021-04-06:
No play