raspicamplayer/camplayer #12: Like to run the app in a portrait orientation, vertically stacked.

blackbear, 2021-03-18:
How hard would it be to set up a configuration to run n 16x9 devices in a 1xn orientation. I have my monitor mounted in portrait on the wall, and if I use any of the stock layouts, it gets horribly stretched.

Glad to help if it's something I can.

blackbear, 2021-03-18:
I have this working, if you give me access I'll submit a pull request.
[image]

raspicamplayer, 2021-03-18:
@blackbear 
Glad to see you got it working.
To issue a pull request, you have to fork this repository, make your changes and then a "pull request" button should popup.

blackbear, 2021-03-18:
Pull request is in.

raspicamplayer, 2021-03-21:
https://github.com/raspicamplayer/camplayer/pull/13

closed 2021-03-21 by raspicamplayer