raspicamplayer/camplayer #14: don't get video rasp 4b

fredjethehulk, 2021-03-21:
Hi,
I am working with raspberry pi 4b 8gb debian buster desktop
I tried to implement my 2 dahua video recorders but i don't get any video if i try the rtsp links in vlc it does work.

i have made my config.ini file like this
[DEVICE1]
channel1_name = Voorkant
channel1.1_url = rtsp://admin:0000000@192.168.0.5:554/cam/realmonitor?channel=1&subtype=0
channel1.2_url = rtsp://admin:0000000@192.168.0.5:554/cam/realmonitor?channel=1&subtype=1

[DEVICE2]
channel2_name = Carport
channel2.1_url = rtsp://admin:0000000@192.168.0.5:554/cam/realmonitor?channel=2&subtype=0
channel2.2_url = rtsp://admin:0000000@192.168.0.5:554/cam/realmonitor?channel=2&subtype=1

[DEVICE3]
channel3_name = Achterkant
channel3.1_url = rtsp://admin:0000000@192.168.0.5:554/cam/realmonitor?channel=3&subtype=0
channel3.2_url = rtsp://admin:0000000@192.168.0.5:554/cam/realmonitor?channel=3&subtype=1

[DEVICE4]
channel4_name = Kot binnen
channel4.1_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=1&subtype=0
channel4.2_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=1&subtype=1

[DEVICE5]
channel5_name = kot deur
channel5.1_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=2&subtype=0
channel5.2_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=2&subtype=1

[DEVICE6]
channel6_name = kot poort
channel6.1_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=3&subtype=0
channel6.2_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=3&subtype=1

[DEVICE7]
channel7_name = kot bolt
channel7.1_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=4&subtype=0
channel7.2_url = rtsp://admin:0000000@192.168.0.190:554/cam/realmonitor?channel=4&subtype=1

[SCREEN1]
layout = 7
window1 = device1,channel1
window2 = device2,channel1
window3 = device3,channel1
window4 = device4,channel1
window5 = device5,channel1
window6 = device6,channel1
window7 = device7,channel1

[ADVANCED]
buffertime = 1000
loglevel = 1
hardwarecheck = 1
icons = 1 
enableaudio = 1
enablevideoosd = 1 
audiovolume = 100 

if i try this **pi@raspberrypi:~ $ ffprobe -hide_banner rtsp://admin:0000000.@192.168.0.5:554/cam/realmonitor?channel=1&subtype=0**
i get this as respond **[rtsp @ 0x1be6a00] method DESCRIBE failed: 404 Not Found
rtsp://admin:0000000.@192.168.0.5:554/cam/realmonitor?channel=1: Server returned 404 Not Found**
 what am i doing wrong?
ps: i am a nooby in this so please help.

this is what i get if i start camplayer
pi@raspberrypi:~ $ camplayer
2021-03-21 20:31:32.167003 - Main - INFO - Starting camplayer version 1.0.0.dev
2021-03-21 20:31:32.167161 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-03-21 20:31:33.926506 - Main - INFO - ********************** SYSTEM INFO **********************
2021-03-21 20:31:33.926865 - Main - INFO - Camplayer version             = 1.0.0.dev
2021-03-21 20:31:33.926993 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
2021-03-21 20:31:33.927137 - Main - INFO - Raspberry Pi SoC              = BCM2711
2021-03-21 20:31:33.927249 - Main - INFO - Raspberry Pi revision         = d03114
2021-03-21 20:31:33.927370 - Main - INFO - Raspberry Pi model name       = 4B 8GB
2021-03-21 20:31:33.927480 - Main - INFO - GPU memory allocation         = 268 MB
2021-03-21 20:31:33.927603 - Main - INFO - Python version                = 3.7.3 (default, Jul 25 2020, 13:03:44)  MB
2021-03-21 20:31:33.927724 - Main - INFO - VLC installed                 = True
2021-03-21 20:31:33.927838 - Main - INFO - pipng installed               = True
2021-03-21 20:31:33.927954 - Main - INFO - ffmpeg installed              = True
2021-03-21 20:31:33.928063 - Main - INFO - *********************************************************
2021-03-21 20:31:33.929579 - Main - INFO - advanced setting overwritten for 'loglevel' is '1'
2021-03-21 20:31:33.929748 - Main - INFO - advanced setting overwritten for 'buffertime' is '1000'
2021-03-21 20:31:33.930888 - Main - INFO - advanced setting overwritten for 'enableaudio' is '1'
2021-03-21 20:31:33.931110 - Main - INFO - advanced setting overwritten for 'enablevideoosd' is '1'
2021-03-21 20:31:33.961429 - Main - INFO - Detected screen resolution for HDMI0 is '1024x600@60Hz'
2021-03-21 20:31:33.988369 - Main - INFO - Using a virtual screen resolution of '1024x600'
2021-03-21 20:31:34.000634 - StreamInfo - INFO - stream properties 'rtsp://xxx:yyy@192.168.0.5:554/cam/realmonitor?channel=1&subtype=0', resolution '1920x1080@0', codec 'hevc', calculated weight '0', valid url '1', has 
2021-03-21 20:31:34.000634 - StreamInfo - INFO --> audio 'True', valid video 'windowed 0 fullscreen 1'
2021-03-21 20:31:34.004249 - StreamInfo - INFO - stream properties 'rtsp://xxx:yyy@192.168.0.5:554/cam/realmonitor?channel=1&subtype=1', resolution '352x288@0', codec 'hevc', calculated weight '0', valid url '1', has au
2021-03-21 20:31:34.004249 - StreamInfo - INFO --> dio 'False', valid video 'windowed 0 fullscreen 1'
2021-03-21 20:31:34.006617 - Screen - INFO - init screen number '1' with a total weight of '0'
2021-03-21 20:31:34.006793 - ScreenManager - INFO - added screen number '1' to display '1' with layout '7' and displaytime '10'

raspicamplayer, 2021-03-21:
Hmm ffprobe seems to fail for whatever reason.
Your streams are hevc/h265 encoded, is that right? If so, please change it to h264 in your camera or DVR, especially for the substream (your stream with subtype=1). If then the detection with ffprobe still fails, remove all your streams with "subtype=0" from your config and disable the hardware check in your config file "hardwarecheck = 0". In any case h265/hevc is not supported for grid type playback at the moment.

fredjethehulk, 2021-03-28:
> Hmm ffprobe seems to fail for whatever reason.
> Your streams are hevc/h265 encoded, is that right? If so, please change it to h264 in your camera or DVR, especially for the substream (your stream [...]

yes in my dvr are they all h264 encoded.
i tried some things and my conclusion is that i only can open 1 camera.
and i have the other issue also that if i close the service that i get a black screen and do not return to desktop screen.

thanks for the help

raspicamplayer, 2021-03-31:
Can you reinstall and retest? I improved ffprobe handling. Be sure to delete the cache folder (/home/pi/.camplayer/cache).