raspicamplayer/camplayer #16: ffprobe exception - TCP issue?

x-rde4, 2021-03-27:
I have an RTSP stream witch works fine with regular omxplayer.
Ex: `omxplayer --avdict rtsp_transport:tcp rtsp://login:pass@ip:557/RVi/1/2 --live`

But when I try to add it to Camplayer I get this terminal output:

```
StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries',
'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate',
'rtsp://login:pass@ip:557/RVi/1/2']' timed out after 10 seconds
```

I guess it's TCP issue, but don't know how to fix it. Can you please give me some advice?

**ffprobe output:**
Comand:
`ffprobe  -hide_banner rtsp://login:pass@ip:557/RVi/1/2`
Output:
```
[rtsp @ 0x169ba00] UDP timeout, retrying with TCP
[rtsp @ 0x169ba00] method PAUSE failed: 551 Option not supported
[rtsp @ 0x169ba00] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size
Consider increasing the value for the 'analyzeduration' and 'probesize' options
```
Comand:
`ffprobe -rtsp_transport tcp -hide_banner rtsp://login:pass@ip:557/RVi/1/2`
Output:
```
Input #0, rtsp, from 'rtsp://login:pass@ip:557/RVi/1/2':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.399333, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuv420p(progressive), 704x480, 15 fps, 15 tbr, 90k tbn, 30 tbc
```

raspicamplayer, 2021-03-31:
@x-rde4 
I've added some tcp/udp handling, can you retest?

x-rde4, 2021-04-01:
@raspicamplayer 
It works! Thanks a lot!

closed 2021-04-01 by x-rde4

andyjmorris, 2021-04-15:
I am having this same issue.  Pi3 B Pelco camera.

StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', 'rtsp://etc....

ffprobe -rtsp_transport tcp -hide_banner rtsp://user:pass@172.16.142.224/stream3
[rtsp @ 0xfb8a50] method DESCRIBE failed: 401 Unauthorized
rtsp://user:pass@172.16.142.224/stream3: Server returned 401 Unauthorized (authorization failed)

Video plays fine on VLC on Windows Computer and another Pi with GUI

Anything I can try?