raspicamplayer/camplayer #20: Cannot stream main/clear RTSP stream from Reolink camera

sjonkeesse, 2021-04-14:
I have two streams coming from my Reolink camera. A `fluent` which is very lightweight stream and a `clear` stream which is a higher quality. I can use the `fluent` stream with the raspicamplayer without any problem. But the `clear` stream results in a "NO LINK". Within the Reolink configuration I tried different resolution, framerates and bitrates. Even the very lowest setting is not working. Both `fluent` and `clear` stream do work with the VLC app and `https://github.com/Anonymousdog/displaycameras`.

[image]

### config.ini
```
[DEVICE1]
channel1_name = Reolink cam
channel1.1_url = rtsp://admin:mypassword@192.168.1.21:554/h264Preview_01_main
#channel1.2_url = rtsp://admin:mypassword@192.168.1.21:554/h264Preview_01_sub

[SCREEN1]
layout = 1
window1 = device1,channel1

[ADVANCED]
enableaudio = 0
```

### Console output when starting the `clear` a.k.a. main stream (not working)
```
pi@raspberrypi:~/.camplayer $ camplayer
2021-04-14 16:22:34.634203 - Main - INFO - Starting camplayer version 1.0.0b4
2021-04-14 16:22:34.634383 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-04-14 16:22:36.460951 - Main - INFO - ********************** SYSTEM INFO **********************
2021-04-14 16:22:36.461343 - Main - INFO - Camplayer version             = 1.0.0b4
2021-04-14 16:22:36.461559 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
2021-04-14 16:22:36.461946 - Main - INFO - Raspberry Pi SoC              = BCM2711
2021-04-14 16:22:36.462136 - Main - INFO - Raspberry Pi revision         = c03112
2021-04-14 16:22:36.462314 - Main - INFO - Raspberry Pi model name       = 4B 4GB
2021-04-14 16:22:36.462499 - Main - INFO - GPU memory allocation         = 256 MB
2021-04-14 16:22:36.462640 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-04-14 16:22:36.462813 - Main - INFO - VLC installed                 = True
2021-04-14 16:22:36.462959 - Main - INFO - pipng installed               = True
2021-04-14 16:22:36.463097 - Main - INFO - ffmpeg installed              = True
2021-04-14 16:22:36.463216 - Main - INFO - *********************************************************
2021-04-14 16:22:36.498781 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2021-04-14 16:22:36.525731 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2021-04-14 16:22:36.540423 - Window - DEBUG - init window with position '0 0 1920 1080', gridindex '[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S
2021-04-14 16:22:36.540423 - Window - DEBUG --> 01_W01'
2021-04-14 16:22:36.544237 - StreamInfo - INFO - stream properties 'rtsp://xxx:yyy@192.168.1.21:554/h264Preview_01_main', resolution '2560x1440@0', codec 'h264', calculated weight '0', valid url '1', has audio 'True', v
2021-04-14 16:22:36.544237 - StreamInfo - INFO --> alid video 'windowed 0 fullscreen 0', force UDP 'False'
2021-04-14 16:22:36.544896 - Screen - INFO - init screen number '1' with a total weight of '0'
2021-04-14 16:22:36.545061 - ScreenManager - INFO - added screen number '1' to display '1' with layout '1' and displaytime '10'
2021-04-14 16:22:36.545529 - BackGroundManager - DEBUG - Loading pipng for display '1' with command '['pipng', '-b', '0', '-l', '-100', '-d', '2', '-h', '-i', '/home/pi/.camplayer/cache/nolink_1x1_1920_1080.png']'
2021-04-14 16:22:36.553442 - BackGroundManager - DEBUG - Loading pipng for display '0' with command '['pipng', '-b', '0', '-l', '1000', '-d', '2', '-i', '-x', '60', '-y', '60', '../resources/icons/icon_loading.png', '../resourc
2021-04-14 16:22:36.553442 - BackGroundManager - DEBUG --> es/icons/icon_paused.png', '../resources/icons/icon_control.png']'
2021-04-14 16:22:36.561062 - ScreenManager - DEBUG - starting all streams for screen '1' on display '1'
2021-04-14 16:22:36.561711 - BackGroundManager - DEBUG - setting background '/home/pi/.camplayer/cache/nolink_1x1_1920_1080.png' visible for display '0
2021-04-14 16:22:36.662542 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-04-14 16:22:36.662695 - ScreenManager - DEBUG - player watchdog
2021-04-14 16:22:51.737255 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-04-14 16:22:51.737732 - ScreenManager - DEBUG - player watchdog
2021-04-14 16:23:06.864102 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-04-14 16:23:06.864577 - ScreenManager - DEBUG - player watchdog
...repeating the last two lines forever...
```

### Console output when starting the `fluent` a.k.a. sub stream (working)
```
pi@raspberrypi:~/.camplayer $ camplayer
2021-04-14 16:37:31.002140 - Main - INFO - Starting camplayer version 1.0.0b4
2021-04-14 16:37:31.002283 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-04-14 16:37:32.792555 - Main - INFO - ********************** SYSTEM INFO **********************
2021-04-14 16:37:32.793403 - Main - INFO - Camplayer version             = 1.0.0b4
2021-04-14 16:37:32.793908 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
2021-04-14 16:37:32.794421 - Main - INFO - Raspberry Pi SoC              = BCM2711
2021-04-14 16:37:32.794923 - Main - INFO - Raspberry Pi revision         = c03112
2021-04-14 16:37:32.795399 - Main - INFO - Raspberry Pi model name       = 4B 4GB
2021-04-14 16:37:32.795704 - Main - INFO - GPU memory allocation         = 256 MB
[... 29 lines cut ...]
2021-04-14 16:37:34.061270 - Window - DEBUG - fetching playstate for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@192.168.1.21:554/h264Preview_01_sub'
2021-04-14 16:37:35.089326 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-04-14 16:37:36.376597 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-04-14 16:37:37.669307 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-04-14 16:37:38.961718 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-04-14 16:37:40.255023 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-04-14 16:37:41.547319 - Window - ERROR - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly after '6' attemps, give up now
2021-04-14 16:37:56.584528 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-04-14 16:37:56.584797 - ScreenManager - DEBUG - player watchdog
2021-04-14 16:37:56.672300 - PIDpool - DEBUG - active OMXplayer PIDs '['2224']'
2021-04-14 16:37:56.680695 - Window - DEBUG - fetching playstate for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@192.168.1.21:554/h264Preview_01_sub'
2021-04-14 16:37:56.704527 - Window - DEBUG - DBus response to command 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01:Duration ' is 'int64 15245000'
2021-04-14 16:37:56.705113 - Window - INFO - stream playstate 'PLAYING' for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@192.168.1.21:554/h264Preview_01_sub'
2021-04-14 16:38:11.747316 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-04-14 16:38:11.747571 - ScreenManager - DEBUG - player watchdog
2021-04-14 16:38:11.840446 - PIDpool - DEBUG - active OMXplayer PIDs '['2224']'
2021-04-14 16:38:11.855828 - Window - DEBUG - fetching playstate for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@192.168.1.21:554/h264Preview_01_sub'
2021-04-14 16:38:11.920827 - Window - DEBUG - DBus response to command 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01:Duration ' is 'int64 30477000'
2021-04-14 16:38:26.960171 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-04-14 16:38:26.960419 - ScreenManager - DEBUG - player watchdog
2021-04-14 16:38:27.052656 - PIDpool - DEBUG - active OMXplayer PIDs '['2224']'
2021-04-14 16:38:27.067877 - Window - DEBUG - fetching playstate for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@192.168.1.21:554/h264Preview_01_sub'
2021-04-14 16:38:27.101292 - Window - DEBUG - DBus response to command 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01:Duration ' is 'int64 45645000'
...repeating last 4 lines forever...
```

raspicamplayer, 2021-04-14:
Your non-working logfile reports "resolution '2560x1440@0", can you post a logfile when the resolution is 1280x720 or lower?
Also helpful would be the output of following command: "ffprobe -rtsp_transport tcp -hide_banner _yourstreamurl_"

sjonkeesse, 2021-04-15:
The non-working logfile was for the 720p camera setting, even though it shows `resolution '2560x1440@0` in the report.

Without any change to the camera settings, ffprobe shows:
```
pi@raspberrypi:~ $ ffprobe -rtsp_transport tcp -hide_banner rtsp://admin:password@192.168.1.21:554/h264Preview_01_main
Input #0, rtsp, from 'rtsp://admin:3b272829R*@192.168.1.21:554/h264Preview_01_main':
  Metadata:
    title           : Session streamed by "preview"
    comment         : h264Preview_01_main
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1280x720, 24.83 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: aac (LC), 16000 Hz, mono, fltp
```

raspicamplayer, 2021-04-15:
@sjonkeesse I forgot to mention one important detail. Camplayer does cache stream information for a faster startup.
Please run "camplayer --rebuild-cache" after changing your camera settings, I think this will solve your issue.

sjonkeesse, 2021-04-16:
@raspicamplayer Thanks, that has indeed solved the problem 🎉

closed 2021-04-16 by sjonkeesse

kathekon, 2023-06-23:
Hi - were you able to get this working with the default Reolink "Clear" settings or did you have to dial the screen resolution way down?  My camera, a Reolink RLC-510A, can only go down to 2304x1296 on "clear" and can only go up to 640x480 on "fluent."    640x480 works, but is super-low resolution for my needs and 2304x1296 does not work (I, like you, tried many different bit rate and frame rate settings).  thanks