raspicamplayer/camplayer #21: Some reolink camera steams go down when connected to camplayer

Pocmanscott, 2021-04-14:
Have 4 Reolink cameras streaming to camplayer on a raspberry pi 4 4GB. Two of the camera streams fail after being connected to camplayer and the cameras have to be reset. When the streams are disconnected from camplayer, they don't go down. 

Using other IP camera streaming software does not recreate this issue.

raspicamplayer, 2021-04-14:
Camplayer can open multiple streams of the same camera for background caching (depending on your config), maybe this causes your cameras to lockup. You can verify this by opening multiple VLC instances for example.

BassixIndy, 2021-05-18:
How do I turn off caching? I could not find any information regarding this in the documentation. Are you talking about setting the "buffertime" to zero? I have an issue with a reolink camera, it works fine for 2 hours straight, but then goes down. Any other tricks I could try?

Pocmanscott, 2021-05-22:
Would also be interested to know if there's a way to disable caching

ChainVTF, 2021-07-23:
I am also seeing issues with Reolink camera streams, using the sub stream 640x480 @ 7fps 512bit.  It either shows artifacts on half the camera view or NO LINK.  No issues when using RTSP via VLC, even with multiple streams to multiple computers.  No issues with the Ubiquiti cameras RTSP feed. Using a Raspberry Pi 4 running latest Camplayer OS.

bigohshin, 2021-11-07:
> I am also seeing issues with Reolink camera streams, using the sub stream 640x480 @ 7fps 512bit. It either shows artifacts on half the camera view or NO LINK. No issues when using RTSP via VLC, even [...]

@ChainVTF I have very similar setup. Can't stream with reolink but works great/fine with ubiquiti. I read BlueIris has problems with "substreams"  on reolinks, possibly Iframe config issues. You have any luck resolving issues with reolink?