raspicamplayer/camplayer #34: GPU memory insufficient

parthi2929, 2021-05-22:
Hi

I tried to install your sw in an RP4 8GB version. Here is my config. 

Neofetch
```
  `.::///+:/-.        --///+//-:``    pi@raspberrypi 
 `+oooooooooooo:   `+oooooooooooo:    -------------- 
  /oooo++//ooooo:  ooooo+//+ooooo.    OS: Raspbian GNU/Linux 10 (buster) armv7l 
  `+ooooooo:-:oo-  +o+::/ooooooo:     Host: Raspberry Pi 4 Model B Rev 1.4 
   `:oooooooo+``    `.oooooooo+-      Kernel: 5.10.17-v7l+ 
     `:++ooo/.        :+ooo+/.`       Uptime: 4 hours, 54 mins 
        ...`  `.----.` ``..           Packages: 1454 (dpkg), 3 (snap) 
     .::::-``:::::::::.`-:::-`        Shell: bash 5.0.3 
    -:::-`   .:::::::-`  `-:::-       Resolution: 1920x1080, 1920x1080 
   `::.  `.--.`  `` `.---.``.::`      DE: LXDE 
       .::::::::`  -::::::::` `       WM: Openbox 
 .::` .:::::::::- `::::::::::``::.    Theme: Adwaita [GTK3] 
-:::` ::::::::::.  ::::::::::.`:::-   Icons: Adwaita [GTK3] 
::::  -::::::::.   `-::::::::  ::::   Terminal: lxterminal 
-::-   .-:::-.``....``.-::-.   -::-   Terminal Font: Monospace 10 
 .. ``       .::::::::.     `..`..    CPU: BCM2711 (4) @ 1.500GHz 
   -:::-`   -::::::::::`  .:::::`     Memory: 975MiB / 7875MiB 
   :::::::` -::::::::::` :::::::.
   .:::::::  -::::::::. ::::::::                              
    `-:::::`   ..--.`   ::::::.
      `...`  `...--..`  `...`
            .::::::::::
             `.-::::-`

```
After installation, when I try to run using below command, I get the error as shown in screenshot below. 

```
camplayer --demo
```
[image]

Kindly help.

mpkogli, 2021-05-26:
See the installation instructions! Either add `gpu_mem=256 ` to /boot/config.txt, or you can also adjust it using the `sudo raspi-config` command. I don't have a Pi in front of me now, but I believe it's under "Performance Options".

closed 2021-07-02 by raspicamplayer

mzsolnay, 2022-01-21:
I am having the same issue, 'GPU memory of 76 insufficient'.
Modified config.txt gpu_mem=256, checked also with raspi-config, after reboot, it shows 256.
Still got the 'insufficient' error.