raspicamplayer/camplayer #43: NO LINK everytime, i dont know what to do..

apo931, 2021-06-25:
Hey peoples,

I dont know what I can do now.. I do all the steps what you said but I still get NO LINK. Why?

My Settings:

```
[DEVICE1]
channel1_name = KAMERA 1
channel1.1_url = rtsp://admin:Bintes2016!@10.0.0.250:554/cam/realmonitor?channel=1&subtype=0
channel1.2_url = rtsp://admin:Bintes2016!@10.0.0.250:554/cam/realmonitor?channel=1&subtype=1

[DEVICE2]
channel1_name = KAMERA 2
channel1.1_url = rtsp://admin:Bintes2016!@10.0.0.250:554/cam/realmontior?channel=2&subtype=0
channel1.2_url =

[DEVICE3]
channel1_name = KAMERA 3
channel1.1_url = rtsp://admin:Bintes2016!@10.0.0.250:554/cam/realmonitor?channel=3&subtype=0
channel1.2_url =

[DEVICE4]
channel1_name = KAMERA 4
channel1.1_url = rtsp://admin:Bintes2016!@10.0.0.250:554/cam/realmonitor?channel=4&subtype=0
channel1.2_url =

[SCREEN1]
layout = 4
windows1 = device1,channel1
windows2 = device2,channel1
windows3 = device3,channel1
windows4 = device4,channel1

```
and after I start it with "camplayer" I got just this in the logs, and the window got 4 screens with each NO LINK.
```
pi@raspberrypi:~ $ camplayer
2021-06-25 13:58:34.116876 - Main - INFO - Starting camplayer version 1.0.0b5
2021-06-25 13:58:34.117023 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-06-25 13:58:35.890205 - Main - INFO - ********************** SYSTEM INFO **********************
2021-06-25 13:58:35.890790 - Main - INFO - Camplayer version             = 1.0.0b5
2021-06-25 13:58:35.891042 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
2021-06-25 13:58:35.891298 - Main - INFO - Raspberry Pi SoC              = BCM2711
2021-06-25 13:58:35.891524 - Main - INFO - Raspberry Pi revision         = d03114
2021-06-25 13:58:35.891749 - Main - INFO - Raspberry Pi model name       = 4B 8GB
2021-06-25 13:58:35.891998 - Main - INFO - GPU memory allocation         = 256 MB
2021-06-25 13:58:35.892258 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-06-25 13:58:35.892504 - Main - INFO - VLC installed                 = False
2021-06-25 13:58:35.892732 - Main - INFO - pipng installed               = True
2021-06-25 13:58:35.892972 - Main - INFO - ffmpeg installed              = True
2021-06-25 13:58:35.893189 - Main - INFO - *********************************************************
2021-06-25 13:58:35.926096 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2021-06-25 13:58:35.952585 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2021-06-25 13:58:35.964029 - Window - DEBUG - init window with position '0 0 960 540', gridindex '[0, 1, 4, 5]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01'
2021-06-25 13:58:35.964612 - Window - DEBUG - init window with position '960 0 1920 540', gridindex '[2, 3, 6, 7]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W02'
2021-06-25 13:58:35.964947 - Window - DEBUG - init window with position '0 540 960 1080', gridindex '[8, 9, 12, 13]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03'
2021-06-25 13:58:35.965260 - Window - DEBUG - init window with position '960 540 1920 1080', gridindex '[10, 11, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W04'
2021-06-25 13:58:35.965633 - Screen - INFO - init screen number '1' with a total weight of '0'
2021-06-25 13:58:35.965900 - ScreenManager - INFO - added screen number '1' to display '1' with layout '4' and displaytime '10'
2021-06-25 13:58:35.968599 - BackGroundManager - DEBUG - Loading pipng for display '1' with command '['pipng', '-b', '0', '-l', '-100', '-d', '2', '-h', '-i', '/home/pi/.camplayer/cache/nolink_2x2_1920_1080.png', '/home/pi/.cam
2021-06-25 13:58:35.968599 - BackGroundManager - DEBUG --> player/cache/nolink_1x1_1920_1080.png']'
2021-06-25 13:58:35.975928 - BackGroundManager - DEBUG - Loading pipng for display '0' with command '['pipng', '-b', '0', '-l', '1000', '-d', '2', '-i', '-x', '60', '-y', '60', '../resources/icons/icon_loading.png', '../resourc
2021-06-25 13:58:35.975928 - BackGroundManager - DEBUG --> es/icons/icon_paused.png', '../resources/icons/icon_control.png']'
2021-06-25 13:58:35.984939 - ScreenManager - DEBUG - starting all streams for screen '1' on display '1'
2021-06-25 13:58:35.985787 - BackGroundManager - DEBUG - setting background '/home/pi/.camplayer/cache/nolink_2x2_1920_1080.png' visible for display '0
2021-06-25 13:58:36.086786 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-06-25 13:58:36.087149 - ScreenManager - DEBUG - player watchdog
2021-06-25 13:58:36.086786 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-06-25 13:58:36.087149 - ScreenManager - DEBUG - player watchdog
2021-06-25 13:58:36.086786 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-06-25 13:58:36.087149 - ScreenManager - DEBUG - player watchdog

```

apo931, 2021-06-28:
Hey,

can anybody help me please? I wanna run this cameras on my monitor at my work. please.

apo931, 2021-07-02:
Anybody here to help me?

raspicamplayer, 2021-07-02:
@apo931 
Probably this typo in your config file:
"window**s**1 = device1,channel1" -> window1 = device1,channel1

apo931, 2021-07-05:
> @apo931
> Probably this typo in your config file:
> "window**s**1 = device1,channel1" -> window1 = device1,channel1

oh man, im so dumb.. now its ok, the script is loading the camera 1 but the rest not.. and the log stops here:

`pi@raspberrypi:~ $ camplayer --rebuild-cache
2021-07-05 11:48:23.444019 - Main - INFO - Starting camplayer version 1.0.0b5
2021-07-05 11:48:23.444156 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-07-05 11:48:25.221128 - Main - INFO - ********************** SYSTEM INFO **********************
2021-07-05 11:48:25.221490 - Main - INFO - Camplayer version             = 1.0.0b5
2021-07-05 11:48:25.221594 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
2021-07-05 11:48:25.221703 - Main - INFO - Raspberry Pi SoC              = BCM2711
2021-07-05 11:48:25.221787 - Main - INFO - Raspberry Pi revision         = d03114
2021-07-05 11:48:25.221871 - Main - INFO - Raspberry Pi model name       = 4B 8GB
2021-07-05 11:48:25.221950 - Main - INFO - GPU memory allocation         = 256 MB
2021-07-05 11:48:25.222066 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-07-05 11:48:25.222182 - Main - INFO - VLC installed                 = True
2021-07-05 11:48:25.222282 - Main - INFO - pipng installed               = True
2021-07-05 11:48:25.222376 - Main - INFO - ffmpeg installed              = True
2021-07-05 11:48:25.222453 - Main - INFO - *********************************************************
2021-07-05 11:48:25.224232 - Main - INFO - advanced setting overwritten for 'loglevel' is '1'
2021-07-05 11:48:25.224465 - Main - INFO - advanced setting overwritten for 'buffertime' is '1000'
2021-07-05 11:48:25.254345 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2021-07-05 11:48:25.278113 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2021-07-05 11:48:45.359956 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', 'rtsp://admin:Bi
2021-07-05 11:48:45.359956 - StreamInfo - ERROR --> ntes2016!@10.0.0.250:554/cam/realmonitor?channel=1&subtype=0', '-rtsp_transport', 'udp']' timed out after 10 seconds
2021-07-05 11:48:45.361481 - StreamInfo - INFO - stream properties 'rtsp://xxx:yyy@10.0.0.250:554/cam/realmonitor?channel=1&subtype=0', resolution '0x0@0', codec '', calculated weight '0', valid url '1', has audio 'Fals
2021-07-05 11:48:45.361481 - StreamInfo - INFO --> e', valid video 'windowed 0 fullscreen 0', force UDP 'False'
2021-07-05 11:48:45.361706 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2021-07-05 11:48:50.615147 - StreamInfo - INFO - stream properties 'rtsp://xxx:yyy@10.0.0.250:554/cam/realmonitor?channel=1&subtype=1', resolution '704x576@20', codec 'h264', calculated weight '8110080', valid url '1',
2021-07-05 11:48:50.615147 - StreamInfo - INFO --> has audio 'False', valid video 'windowed 1 fullscreen 1', force UDP 'False'
2021-07-05 11:48:50.615650 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2021-07-05 11:48:50.617927 - Screen - INFO - init screen number '1' with a total weight of '8110080'
2021-07-05 11:48:50.618132 - ScreenManager - INFO - added screen number '1' to display '1' with layout '4' and displaytime '10'
2021-07-05 11:48:51.675536 - Window - INFO - starting stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@10.0.0.250:554/cam/realmonitor?channel=1&subtype=1' with resolution '704x576' and weight '8
2021-07-05 11:48:51.675536 - Window - INFO --> 110080' in a window '960x540'
2021-07-05 11:48:52.773879 - Window - INFO - stream playstate 'INIT2' for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@10.0.0.250:554/cam/realmonitor?channel=1&subtype=1'
2021-07-05 11:48:53.833691 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-07-05 11:48:55.126564 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-07-05 11:48:56.398124 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' is not responding correctly, retrying within 250ms
2021-07-05 11:48:56.859260 - Window - INFO - stream playstate 'PLAYING' for stream 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01' 'rtsp://xxx:yyy@10.0.0.250:554/cam/realmonitor?channel=1&subtype=1'

`

sevenseal, 2021-07-06:
Change the rest of "windows#" to window# ?

Robby73, 2021-07-20:
Also where you have    

[DEVICE4]
channel4_name = KAMERA 4
channel4.1_url = rtsp://xxxx:xxxx@xxxx:554/cam/realmonitor?channel=4&subtype=0

that should be channel1.1_url   not channel4.1_url.    

its channel 1 on device 4 i believe..     
Thats the only way it would work for me anyway

mzsolnay, 2022-01-23:
I am having similar issues ... NO LINK

this is my ini:

[DEVICE1]
channel1_name = Device1
channel1.1_url = rstp://admin:admin@192.168.0.101:80/0
channel1.2_url = rstp://admin:admin@192.168.0.101:80/1

[DEVICE2]
channel1_name = Device2
channel1.1_url = rstp://admin@admin@192.168.0.102:80/0
channel1.2_url = rstp://admin:admin@192.168.0.102:80/1

[SCREEN1]
layout = 4
window1 = device1,channel1
window2 = device1,channel1
window3 = device2,channel1
window4 = device2,channel1

And this is log:

2022-01-23 07:55:08.263082 - Main - INFO - Starting camplayer version 1.0.0b5
2022-01-23 07:55:08.264177 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2022-01-23 07:55:10.011421 - Main - INFO - ********************** SYSTEM INFO **********************
2022-01-23 07:55:10.011765 - Main - INFO - Camplayer version             = 1.0.0b5
2022-01-23 07:55:10.011867 - Main - INFO - Operating system              = Linux raspberrypi 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux
2022-01-23 07:55:10.011960 - Main - INFO - Raspberry Pi SoC              = BCM2835
2022-01-23 07:55:10.012033 - Main - INFO - Raspberry Pi revision         = a020d3
2022-01-23 07:55:10.012109 - Main - INFO - Raspberry Pi model name       = 3B+
2022-01-23 07:55:10.012184 - Main - INFO - GPU memory allocation         = 256 MB
2022-01-23 07:55:10.012275 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2022-01-23 07:55:10.012373 - Main - INFO - VLC installed                 = True
2022-01-23 07:55:10.012454 - Main - INFO - pipng installed               = True
2022-01-23 07:55:10.012533 - Main - INFO - ffmpeg installed              = True
2022-01-23 07:55:10.012600 - Main - INFO - *********************************************************
2022-01-23 07:55:10.121390 - Main - INFO - Detected screen resolution for HDMI0 is '1920x1080@60Hz'
2022-01-23 07:55:10.121712 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2022-01-23 07:55:10.129201 - Window - DEBUG - init window with position '0 0 960 540', gridindex '[0, 1, 4, 5]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01'
2022-01-23 07:55:10.131006 - Window - DEBUG - init window with position '960 0 1920 540', gridindex '[2, 3, 6, 7]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W02'
2022-01-23 07:55:10.133908 - Window - DEBUG - init window with position '0 540 960 1080', gridindex '[8, 9, 12, 13]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03'
2022-01-23 07:55:10.139810 - Window - DEBUG - init window with position '960 540 1920 1080', gridindex '[10, 11, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W04'
2022-01-23 07:55:10.143613 - StreamInfo - INFO - stream properties 'rstp://xxx:yyy@192.168.0.101:80/0', resolution '0x0@0', codec '', calculated weight '0', valid url '0', has audio 'False', valid video 'windowed 0 full
2022-01-23 07:55:10.143613 - StreamInfo - INFO --> screen 0', force UDP 'False'
2022-01-23 07:55:10.145528 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2022-01-23 07:55:10.147533 - StreamInfo - INFO - stream properties 'rstp://xxx:yyy@192.168.0.101:80/1', resolution '0x0@0', codec '', calculated weight '0', valid url '0', has audio 'False', valid video 'windowed 0 full
2022-01-23 07:55:10.147533 - StreamInfo - INFO --> screen 0', force UDP 'False'
2022-01-23 07:55:10.151048 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2022-01-23 07:55:10.153718 - StreamInfo - INFO - stream properties 'rstp://xxx:yyy@192.168.0.101:80/0', resolution '0x0@0', codec '', calculated weight '0', valid url '0', has audio 'False', valid video 'windowed 0 full
2022-01-23 07:55:10.153718 - StreamInfo - INFO --> screen 0', force UDP 'False'
2022-01-23 07:55:10.156139 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2022-01-23 07:55:10.158730 - StreamInfo - INFO - stream properties 'rstp://xxx:yyy@192.168.0.101:80/1', resolution '0x0@0', codec '', calculated weight '0', valid url '0', has audio 'False', valid video 'windowed 0 full
2022-01-23 07:55:10.158730 - StreamInfo - INFO --> screen 0', force UDP 'False'
2022-01-23 07:55:10.161478 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!

Dutchy-79, 2023-07-10:
Same problem, sometimes it runs for days other times it's stopping several times a hour.

theyetter, 2023-07-10:
@mzsolnay - It's rtsp not rstp.
@Dutchy-79  - Your cameras may only send resolution with the keyframes in which case if it loses sync and has to reprobe, if the reprobe period times out before it can get the resolution, then the camera will blank.  Different cameras handle this in different ways so it may take some trial and error or tech support or help on forums for other camera related products to figure out proper settings for your camera - otherwise you may have to hack the python code to extend the timeout.  And this is just one of several possible causes of the issue.

Dutchy-79, 2023-07-11:
Ty but no idea where to start.

I tried camera link with h264 and a resolution behind it, h264 only and just the link. All same issue.

theyetter, 2023-07-12:
@Dutchy-79 -  This is a likely abandoned, definitely unsupported open source project - from what you've said you've tried, you'd likely be much happier using a commercial product like a hardware nvr or blue iris software.  This is the type of situation where you could spend 100 hours on it and get nowhere.  Some are willing to do that for the thrill of the experience but most are not.  You never even mentioned what kind of cameras you had even after I said there could be differences even within the same brand so the road before you to make it work could be very, very long.

Dutchy-79, 2023-07-12:
Ty, I didn't mention I'm using eziv c8c because I thought you don't know all details about the kinds of cameras out there. My bad.

Dutchy-79, 2023-07-12:
Also I'm running only 1 camera, playing on the raspberry pi 7" touchscreen. So buying a dedicated system with monitor is not a option for the amount of money imo.

theyetter, 2023-07-12:
@Dutchy-79 - you will never likely get a stable steady stream from a camera like that.

theyetter, 2023-07-12:
And if you're only using it to watch 1 camera, just use vlc player or omxplayer  - this script is just a frontend for those apps and any advantages of using it are lost if only using it for 1 camera.  You could also use ffplay.

Dutchy-79, 2023-07-13:
Ty very much, I couldn't find that only camplayer.