raspicamplayer/camplayer #44: Config Menu

chenolips, 2021-07-09:
How do I get back from camplayer to the configuration menu?

Sorry for the dumb question, I'm realy new to this.

And it's an awesome project.

bob454522, 2021-07-15:
Same issue here,  i was able to get the config screen initally on first install/boot (of the cam OS , not camplayer app).

then after adding my camera, i can never get back to the Configuration menu again.  On reboot, it starts directly into the camera cyceling.  (i have tried q and Q and every possible key on the keyboard,  as well as trying different types of usb keyboards, just in case)  -  im stuck in the camera display cycle.

great app / potential too! thanks for all the hard work!

cybermaus, 2021-07-15:
I do not have the OS version, but as a guess: Can you SSH into the box with PuTTY? Maybe you need to first activate SSH by taking out the SD card into Windows and an empty text file /boot/ssh.txt

chenolips, 2021-07-15:
Same here. Every possible keyboard ans.configurations.

But I've finally made a way back to the config menu.

When you finish the camplayer wizard, set a password.

You'll be prompted to enter it when turninf off and on the raspberry.

Just enter the the password wrong 5 times and, voilà, you're back to the menu.

I don't think it's the right way. But io if works, it works.

donkurt, 2021-07-18:
The startup screen (OS version) shows the ways to get back to the configuration menu.  Try typing 1 2 3 4 using the keyboard, and viola, the menu appears.  Simple instructions provided, simple solution to follow :)

chenolips, 2021-07-18:
> The startup screen (OS version) shows the ways to get back to the configuration menu. Try typing 1 2 3 4 using the keyboard, and viola, the menu appears. Simple instructions provided, simple solution to follow :)

---

I really think that it was obvious enough that I've already tried  this.

But I'll make it simple for you: for me, at least, it doesn't work at all. Tried with usb and wireless keyboards.