raspicamplayer/camplayer #45: No auto-resolution

cybermaus, 2021-07-13:
Hi.

Very nice product. Installed and got it working with little effort.

When I set `streamquality = 1` (automatic), should camplayer not automatically switch from substream to main stream when I focus on a single camera (by Enter or keys 1~4)

It does work when I press up or down, so the streams are properly defined
(though it is a little slow in switching)

Thanks

```
#devices removed as for sensitive information

[SCREEN1]
layout = 4
window1 = device1,channel1
window2 = device2,channel1
window3 = device3,channel1
window4 = device4,channel1

[SCREEN2]
layout = 4
window1 = device5,channel1
window2 = device6,channel1
window3 = device7,channel1
window4 = device8,channel1

[ADVANCED]
buffertime = 100
showtime =0
icons = 1
streamquality = 1
enablevideoosd = 1
```

cybermaus, 2021-07-18:
Closing this as, now I have studied the source a lot better, I no longer have above question.

closed 2021-07-18 by cybermaus

dorinclisu, 2023-04-20:
I have the same question, I don't understand the behavior of this parameter.