raspicamplayer/camplayer #51: closing pipng causes black screen and gpu hangups on Raspi 4

cybermaus, 2021-07-19:
After a lot of fighting it, where it all worked stable on a 3B but badly on a 4B, I finally traced it to pipng
In fact, I can get the the system to hang by just manually running the same pipng from command line.
```
pipng -b 0 -l -100 -d 2 -h -i /home/pi/.camplayer/cache/nolink_1x1_1280_1024.png
pipng -b 0 -l 1000 -d 2 -i -x 60 -y 60 ../resources/icons/icon_loading.png 
```
Sometimes it ends without hanging, but usually the 2nd and definitely the 3rd run will kill it.
And subsequently, also any 2nd or 3rd camplayer run. And tvservice also will hang after a pipng run.
To reset it, often rebooting is not enough, it has to be power cycled.

First hint something is wrong is that using `ps -ef` both pipng processes persist when you quit camplayer.

Workaround, add this to config.ini:
```
[ADVANCED]
backgroundmode = 3
icons = 0
```

On a 3B I have not seen the issue. On a 3B+, I believe I did also see it, but when I was playing with that I was still new to the entire product, so my memory of problems with that may be other causes. And now I do not have the 3B+ available anymore.
On a 4B, it *definitely* hangs, and (for me) very consistently so. And adding the above to stop using pipng also works very well.

And side-node: Original branded RasPi4 power supply and no messages from the power watchdog.

cybermaus, 2021-07-25:
Addendum: It seems to be only the icons that cause this.
If I remove icons, but leave background as is, it seems to be more stable.
```
[ADVANCED]
#backgroundmode = 3
icons = 0
```
Which is good. Because seeing the X or console reappear every time you switch channels is not so nice. The icons are less important.

tuxthepenguin84, 2022-05-24:
Thanks, this fixed some oddity on my Pi4 I could not narrow down until this post. My GPU would hang and nothing would fix it until I rebooted.