raspicamplayer/camplayer #64: Can't play RTSP streams

kareemtawab, 2021-08-16:
Hi

I am using camplayer on raspbian lite. These are my configs:

```
[DEVICE1]
channel1_name = 1
channel1.1_url = rtsp://192.168.1.11:10554/tcp/av0_1
channel1.2_url = rtsp://192.168.1.11:10554/tcp/av0_1

[DEVICE2]
channel1_name = 2
channel1.1_url = rtsp://192.168.1.15:10554/tcp/av0_1
channel1.2_url = rtsp://192.168.1.15:10554/tcp/av0_1

[DEVICE3]
channel1_name = 3
channel1.1_url = rtsp://192.168.1.13:10554/tcp/av0_1
channel1.2_url = rtsp://192.168.1.13:10554/tcp/av0_1

[SCREEN1]
layout = 4
window1 = device1,channel1
window2 = device2,channel1
window3 = device3,channel1
window4 = device3,channel1

[ADVANCED]
showtime = 0
screenwidth = 640
screenheight = 480
enablehevc = 0
streamquality = 0
loglevel = 0
```

and on starting camplayer I get:

```
2021-08-16 09:24:29.587320 - Main - INFO - Starting camplayer version 1.0.0b5
2021-08-16 09:24:29.587539 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-08-16 09:24:31.365863 - Main - INFO - ********************** SYSTEM INFO **********************
2021-08-16 09:24:31.366296 - Main - INFO - Camplayer version             = 1.0.0b5
2021-08-16 09:24:31.366487 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
2021-08-16 09:24:31.366738 - Main - INFO - Raspberry Pi SoC              = BCM2835
2021-08-16 09:24:31.366862 - Main - INFO - Raspberry Pi revision         = a02082
2021-08-16 09:24:31.366988 - Main - INFO - Raspberry Pi model name       = 3B
2021-08-16 09:24:31.367117 - Main - INFO - GPU memory allocation         = 256 MB
2021-08-16 09:24:31.367262 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-08-16 09:24:31.367402 - Main - INFO - VLC installed                 = False
2021-08-16 09:24:31.367529 - Main - INFO - pipng installed               = True
2021-08-16 09:24:31.367649 - Main - INFO - ffmpeg installed              = True
2021-08-16 09:24:31.367760 - Main - INFO - *********************************************************
2021-08-16 09:24:31.369543 - Main - INFO - advanced setting overwritten for 'screenwidth' is '640'
2021-08-16 09:24:31.370278 - Main - INFO - advanced setting overwritten for 'screenheight' is '480'
2021-08-16 09:24:31.372405 - Main - INFO - advanced setting overwritten for 'showtime' is '0'
2021-08-16 09:24:31.372709 - Main - INFO - advanced setting overwritten for 'streamquality' is '0'
2021-08-16 09:24:31.373034 - Main - INFO - advanced setting overwritten for 'enablehevc' is '0'
2021-08-16 09:24:31.373883 - Main - INFO - Using a virtual screen resolution of '640x480'
2021-08-16 09:24:31.384710 - Window - DEBUG - init window with position '0 0 320 240', gridindex '[0, 1, 4, 5]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01'
2021-08-16 09:24:31.385346 - Window - DEBUG - init window with position '320 0 640 240', gridindex '[2, 3, 6, 7]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W02'
2021-08-16 09:24:31.385795 - Window - DEBUG - init window with position '0 240 320 480', gridindex '[8, 9, 12, 13]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03'
2021-08-16 09:24:31.386256 - Window - DEBUG - init window with position '320 240 640 480', gridindex '[10, 11, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W04'
2021-08-16 09:24:44.999542 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', 'rtsp://192.168.
2021-08-16 09:24:44.999542 - StreamInfo - ERROR --> 1.11:10554/tcp/av0_1', '-rtsp_transport', 'udp']' timed out after 10 seconds
2021-08-16 09:24:45.000632 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/tcp/av0_1', resolution '0x0@0', codec '', calculated weight '0', valid url '1', has audio 'False', valid video 'windowed 0 fu
2021-08-16 09:24:45.000632 - StreamInfo - INFO --> llscreen 0', force UDP 'False'
2021-08-16 09:24:45.000952 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
```

No matter what, it tries every 10 seconds and then 'NO LINK'. My thinking is that it tries to start the stream using "RTSP over UDP" instead of "RTSP" as in the camplayer OS. I can see a hint of it in the second error line ('udp'). Could be it only an argument to the player?

cybermaus, 2021-08-16:
If you directly run command `ffprobe -v error -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -rtsp_transport udp rtsp://192.168.1.11:10554/tcp/av0_1` what do you get?

Also try without the `-v error`
And replacing to ` -rtsp_transport tcp`

kareemtawab, 2021-08-16:
Running the command without `-v error` and with `-rtsp_transport tcp` gives:
```
ffprobe version 4.1.6-1~deb10u1+rpt2 Copyright (c) 2007-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt2' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-vout-drm --enable-v4l2-request --enable-libudev --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt2' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-vout-drm --enable-v4l2-request --enable-libudev --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt2' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-vout-drm --enable-v4l2-request --enable-libudev --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt2' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi - [... cut ...]
height=480
avg_frame_rate=15/1
bit_rate=N/A
max_bit_rate=N/A
[/STREAM]
[STREAM]
codec_name=pcm_alaw
codec_type=audio
avg_frame_rate=0/0
bit_rate=64000
max_bit_rate=N/A
[/STREAM]
```

Running the command with `-rtsp_transport tcp` only gives:
```
[STREAM]
codec_name=h264
codec_type=video
width=640
height=480
avg_frame_rate=15/1
bit_rate=N/A
max_bit_rate=N/A
[/STREAM]
[STREAM]
codec_name=pcm_alaw
codec_type=audio
avg_frame_rate=0/0
bit_rate=64000
max_bit_rate=N/A
[/STREAM]
```

I guess it is streaming.

cybermaus, 2021-08-16:
So from the fact that your manual ffprobe with udp also gives all empty values, and the one with tcp works, I think you suggestion that it may be due to use of udp seems correct. But how to fix it, I do not know. 

For me, when running omxplayer manually, I had the same issue, I had to use tcp for some cameras, which is why I was so happy to see this software did it automatically correct.

kareemtawab, 2021-08-16:
I have tried the OS before that and while I was adding the URLs I selected to preview the feed before adding the cameras and I was able to see the preview correctly at the corner of the screen. However, after adding the camera it was not able to get the feed on the main screen.

I will look into the code to find the argument, if it was that easy enough. I can probably change it.

cybermaus, 2021-08-16:
Reading the source in streaminfo, I see how to hack, but I also see it should have already tried tcp.
Can you try to delete and rebuild the cache?  `camplayer --help`  should show how, but alternatively you can just delete  `~/.camplayer/cache/streaminfo`

kareemtawab, 2021-08-16:
Deleting `~/.camplayer/cache/streaminfo` not do anything. I'm out of ideas.

cybermaus, 2021-08-17:
It is weird that if you try manually with tcp it works, but the code seems primed to also try tcp first and only fallback to udp. Also your dump above actually claims it found `force UDP false`. But yet all stream values are empty/0

Does it even populate the cache/streaminfo file? Can you show the content?

kareemtawab, 2021-08-18:
I freshly ran Raspbian with a fresh install of camplayer. Here, I was using this free RTSP link [rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov](url) and some other local files and indeed it includes the info in the cache/streaminfo file.

```
{
    "file:///usr/local/share/camplayer/resources/video/elephants_dream_120p.mp4": {
        "codec_name": "h264",
        "height": 120,
        "width": 214,
        "framerate": 24.0,
        "audio": true,
        "force_udp": false
    },
    "file:///usr/local/share/camplayer/resources/video/elephants_dream_360p.mp4": {
        "codec_name": "h264",
        "height": 360,
        "width": 640,
        "framerate": 24.0,
        "audio": true,
        "force_udp": false
    },
    "file:///usr/local/share/camplayer/resources/video/elephants_dream_1080p.mp4": {
        "codec_name": "h264",
        "height": 1080,
        "width": 1920,
        "framerate": 24.0,
        "audio": true,
        "force_udp": false
    },
    "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov": {
        "codec_name": "h264",
        "height": 160,
        "width": 240,
        "framerate": 24.0,
        "audio": true,
        "force_udp": false
    }
}
```
However, it does not even create a cache/streaminfo file if I include all of my other camera feed URLs only in the config. It will only include data for other free test RTSP URLs or local files.

cybermaus, 2021-08-18:
I highly doubt it is a problem with omxplayer. The cache/streaminfo content is generated with `ffprobe` and next camplayer uses that to determine how to best run omxplayer.

Both the log in your first post, and the fact cache/streaminfo stays blank, means ` ffprobe` already have a problem, and at that point omxplayer has not been called yet. Something is up with your rtsp stream or url.

The only weird thing is, when you did the manual `ffprobe`  it worked.

I think it is something with the URL. 
And the only think I see unusual with that is the port: `:10544`
I wonder if the `:` is trowing off the system. Can you try some variations of the URL. Also try them with manual ` ffprobe` 

Can you move the RTSP port to 554 so you can leave of the `:10544` completely?

` ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate "rtsp://192.168.1.11:10554/tcp/av0_1" `
` ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -- rtsp://192.168.1.11:10554/tcp/av0_1 `
` ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -- "rtsp://192.168.1.11:10554/tcp/av0_1" `
` ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate rtsp://192.168.1.11%3A10554/tcp/av0_1 `

To be honest, I am randomly trying a bit with above. But I do believe internally camplayer should be using the `--` in both the ffprobe and the omxplayer calls. I had to tweak my own version to add those to be able to play youtube video's

cybermaus, 2021-08-18:
Oh, and also, delete the 2nd stream. Rather silly to offer the same quality stream twice:

```[DEVICE1]
channel1_name = 1
channel1.1_url = rtsp://192.168.1.11:10554/tcp/av0_1
c̶h̶a̶n̶n̶e̶l̶1̶.̶2̶_̶u̶r̶l̶ ̶=̶ ̶r̶t̶s̶p̶:̶/̶/̶1̶9̶2̶.̶1̶6̶8̶.̶1̶.̶1̶1̶:̶1̶0̶5̶5̶4̶/̶t̶c̶p̶/̶a̶v̶0̶_̶1̶
```

kareemtawab, 2021-08-18:
> Oh, and also, delete the 2nd stream. Rather silly to offer the same quality stream twice:
> 
> ```
> channel1_name = 1
> channel1.1_url = rtsp://192.168.1.11:10554/tcp/av0_1
> c̶h̶a̶n̶n̶e̶l̶1̶.̶2̶_̶u̶r̶l̶ ̶=̶ ̶r̶t̶s̶p̶:̶/̶/̶1̶9̶2̶.̶1̶6̶8̶.̶1̶.̶1̶1̶:̶1̶0̶5̶5̶4̶/̶t̶c̶p̶/̶a̶v̶0̶_̶1̶
> ```

I was using the demo config example and it had multiple streams of different qualities, so I thought I'll populate them with the only stream quality I need as camplayer shuffles between the views/windows and the streams. Surely, I'll remove all links except one per channel. Thanks.

cybermaus, 2021-08-18:
Having the same stream in there twice would cause it to ffprobe it twice, which I do not think is the cause of the problem, but it surely will not help either.

Anyway, any luck with the suggestion to play with how to write the URL and its port number?

kareemtawab, 2021-08-18:
> I highly doubt it is a problem with omxplayer. The cache/streaminfo content is generated with `ffprobe` and next camplayer uses that to determine how to best run omxplayer.
>
> Both the log in your [...]

From these:

`ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate "rtsp://192.168.1.11:10554/tcp/av0_1"`
`ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -- rtsp://192.168.1.11:10554/tcp/av0_1`
`ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -- "rtsp://192.168.1.11:10554/tcp/av0_1"`
`ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate rtsp://192.168.1.11%3A10554/tcp/av0_1`

I got this one working, only after changing `tcp` to `udp` in the URL:

`ffprobe -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -- rtsp://192.168.1.11:10554/tcp/av0_1`

and outputs:

```
ffprobe version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2007-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libas [... cut ...]
Regarding the port number, I tried many different URLs and I searched the internet but I couldn't find any other than these:

`rtsp://xxx.xxx.xxx.xxx:10554/tcp/av0_1` - 640x480
`rtsp://xxx.xxx.xxx.xxx:10554/udp/av0_1` - 640x480
`rtsp://xxx.xxx.xxx.xxx:10554/tcp/av0_0` - 1280x720
`rtsp://xxx.xxx.xxx.xxx:10554/udp/av0_0` - 1280x720

and I couldn't find a way of predicting it. I can not find anything for the port 10554 for this camera. Camera port settings below as well as its model number. 

[image]

[image]

cybermaus, 2021-08-19:
I overlooked that /tcp/ in the URL, thanks.
Also, since you do have 2 qualities, maybe run all tests on both 0_0 and 0_1 streams, just in case the camera behaves different on the lower resolution substream.

So from earlier report, this works:
(the `..........`  is of course that whole preample, but honestly, you can do with or without)
`ffprobe .......... -rtsp_transport tcp rtsp://192.168.1.11:10554/tcp/av0_0` 
And this also works
`ffprobe .......... -- rtsp://192.168.1.11:10554/udp/av0_0` 

But this one did not work? 
`ffprobe .......... rtsp://192.168.1.11:10554/udp/av0_0` 

How about these:
`ffprobe .......... -rtsp_transport udp rtsp://192.168.1.11:10554/udp/av0_0` 
`ffprobe .......... -rtsp_transport udp -- rtsp://192.168.1.11:10554/udp/av0_0` 

And of course, this one also works:
`ffprobe .......... -rtsp_transport tcp -- rtsp://192.168.1.11:10554/tcp/av0_0` 

BTW: you once mentioned it worked in VLC. I trust you tried both udp and tcp and both 0_0 and 0_1 variants in VLC?

--------

BTW: since you do have a 2nd stream with different qualities, you could add back in that 2nd stream.
But lets do that later, if/when we got it to work

```
channel1_name = 1
channel1.1_url = rtsp://192.168.1.11:10554/tcp/av0_0
channel1.2_url = rtsp://192.168.1.11:10554/tcp/av0_1
```

kareemtawab, 2021-08-19:
### ffprobe:

All of the possible trials, so far for `...../av0_0` (1280x720):

`ffprobe -rtsp_transport tcp rtsp://192.168.1.11:10554/tcp/av0_0`
`ffprobe -rtsp_transport tcp -- rtsp://192.168.1.11:10554/tcp/av0_0`
`ffprobe -rtsp_transport udp rtsp://192.168.1.11:10554/tcp/av0_0`
`ffprobe -rtsp_transport udp -- rtsp://192.168.1.11:10554/tcp/av0_0`
`ffprobe rtsp://192.168.1.11:10554/tcp/av0_0`
`ffprobe rtsp://192.168.1.11:10554/udp/av0_0`

They all output:

```
  Metadata:
    title           : streamed by the Zmodo RTSP server
  Duration: N/A, start: 0.040000, bitrate: N/A
    Stream #0:0: Video: h264 (Baseline), yuv420p(progressive), 1280x720, 25 fps, 25 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: pcm_alaw, 8000 Hz, 1 channels, s16, 64 kb/s
```

And for `...../av0_1` (640x480):

`ffprobe -rtsp_transport tcp rtsp://192.168.1.11:10554/tcp/av0_1`
`ffprobe -rtsp_transport tcp -- rtsp://192.168.1.11:10554/tcp/av0_1`
`ffprobe -rtsp_transport udp rtsp://192.168.1.11:10554/tcp/av0_1`
`ffprobe -rtsp_transport udp -- rtsp://192.168.1.11:10554/tcp/av0_1`
`ffprobe rtsp://192.168.1.11:10554/tcp/av0_1`
`ffprobe rtsp://192.168.1.11:10554/udp/av0_1`

They all output:

```
  Metadata:
    title           : streamed by the Zmodo RTSP server
  Duration: N/A, start: 0.100000, bitrate: N/A
    Stream #0:0: Video: h264 (Baseline), yuv420p(progressive), 640x480, 10 fps, 10 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: pcm_alaw, 8000 Hz, 1 channels, s16, 64 kb/s
```

### Testing on VLC:

from a Windows 10 machine:

[image]

I also tried to run them on VLC Desktop from the same Raspbian (no HW acceleration here I believe) and they are all working. I just can't take a snapshot from VNC viewer but here is a photo of VLC running `rtsp://192.168.1.11:10554/udp/av0_0` at 1280x720:

[image]

kareemtawab, 2021-08-19:
I'm still suspecting omxplayer as it recognises the stream metadata but does not play it on screen for some reason and wishes me a nice day. Running both:

`omxplayer -- rtsp://192.168.1.11:10554/udp/av0_1`
`omxplayer -- rtsp://192.168.1.11:10554/udp/av0_0`

give:

```
Video codec omx-h264 width 640 height 480 profile 66 fps 10.000000
Audio codec pcm_alaw channels 1 samplerate 8000 bitspersample 8
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 640x480@10.00 interlace:0 deinterlace:0 anaglyph:0 par:0.94 display:0 layer:0 alpha:255 aspectMode:0
have a nice day ;)
```
and
```
Video codec omx-h264 width 1280 height 720 profile 66 fps 25.000000
Audio codec pcm_alaw channels 1 samplerate 8000 bitspersample 8
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:0.94 display:0 layer:0 alpha:255 aspectMode:0
have a nice day ;)
```

I tried all of the possible URLs I wrote in the previous post and it just does not want to play any of them.

To roll things out, I tried many public free URLs and it plays most of them and other there were just connectivity problems and not problems with playback.

`rtsp://freja.hiof.no:1935/rtplive/_definst_/hessdalen02.stream`
`rtsp://freja.hiof.no:1935/rtplive/_definst_/hessdalen03.stream`

give:

```
Video codec omx-h264 width 1280 height 720 profile 100 fps 59.940060
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1280x720@59.94 interlace:0 deinterlace:0 anaglyph:0 par:0.94 display:0 layer:0 alpha:255 aspectMode:0
have a nice day ;)
```
and
```
Video codec omx-h264 width 1280 height 720 profile 100 fps 59.940060
Audio codec aac channels 2 samplerate 48000 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1280x720@59.94 interlace:0 deinterlace:0 anaglyph:0 par:0.94 display:0 layer:0 alpha:255 aspectMode:0
have a nice day ;)
```
respectively and plays them successfully on screen.

cybermaus, 2021-08-19:
So, it is working quite well from command line. But not at all from the program.
Even though the program uses the same ffprobe.

Lets go deeper. In /usr/local/share/camplayer/camplayer/streaminfo.py 
Change:
```
                try:
                    ffprobe_args = ['ffprobe', '-v', 'error', '-show_entries',
                                    'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate',
                                    self.url]

                    if self.url.startswith('rtsp://'):
                        ffprobe_args.extend(['-rtsp_transport', transport])

                    # Invoke ffprobe, 20s timeout required for pi zero
                    streams = subprocess.check_output(ffprobe_args, universal_newlines=True, timeout=10,
                                                      stderr=subprocess.STDOUT).split("[STREAM]")
```
to 
```
                try:
                    ffprobe_args = ['ffprobe', '-v', 'error', '-show_entries',
                                    'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate',
                                    '--', self.url]

                    if self.url.startswith('rtsp://'):
                        ffprobe_args.extend(['-rtsp_transport', transport])

                    print(str(ffprobe_args))

                    # Invoke ffprobe, 20s timeout required for pi zero
                    streams = subprocess.check_output(ffprobe_args, universal_newlines=True, timeout=10,
                                                      stderr=subprocess.STDOUT).split("[STREAM]")

                    print(str(streams))

```
So add both prints and the `'--',`
Run and show output.
I wrote those prints from memory, as I did the location of the file, so look around a little.

kareemtawab, 2021-08-19:
Displays 4 blank streams and outputs:

```
2021-08-19 09:24:14.753201 - Main - INFO - Starting camplayer version 1.0.0b5
2021-08-19 09:24:14.753423 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-08-19 09:24:16.540643 - Main - INFO - ********************** SYSTEM INFO **********************
2021-08-19 09:24:16.540999 - Main - INFO - Camplayer version             = 1.0.0b5
2021-08-19 09:24:16.541127 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
2021-08-19 09:24:16.541263 - Main - INFO - Raspberry Pi SoC              = BCM2835
2021-08-19 09:24:16.541376 - Main - INFO - Raspberry Pi revision         = a02082
2021-08-19 09:24:16.541496 - Main - INFO - Raspberry Pi model name       = 3B
2021-08-19 09:24:16.541619 - Main - INFO - GPU memory allocation         = 256 MB
2021-08-19 09:24:16.541760 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-08-19 09:24:16.541896 - Main - INFO - VLC installed                 = True
2021-08-19 09:24:16.542015 - Main - INFO - pipng installed               = True
2021-08-19 09:24:16.542126 - Main - INFO - ffmpeg installed              = True
2021-08-19 09:24:16.542234 - Main - INFO - *********************************************************
2021-08-19 09:24:16.576151 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2021-08-19 09:24:16.576572 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2021-08-19 09:24:16.585328 - Window - DEBUG - init window with position '0 0 960 540', gridindex '[0, 1, 4, 5]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01'
2021-08-19 09:24:16.585847 - Window - DEBUG - init window with position '960 0 1920 540', gridindex '[2, 3, 6, 7]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W02'
2021-08-19 09:24:16.586202 - Window - DEBUG - init window with position '0 540 960 1080', gridindex '[8, 9, 12, 13]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03'
2021-08-19 09:24:16.586533 - Window - DEBUG - init window with position '960 540 1920 1080', gridindex '[10, 11, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W04'
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://192.168.1.11:10554/udp/av0_0', '-rtsp_transport', 'tcp']
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://192.168.1.11:10554/udp/av0_0', '-rtsp_transport', 'udp']
2021-08-19 09:24:17.477356 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://19
2021-08-19 09:24:17.477356 - StreamInfo - ERROR --> 2.168.1.11:10554/udp/av0_0', '-rtsp_transport', 'udp']' returned non-zero exit status 1.
2021-08-19 09:24:17.478170 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/udp/av0_0', resolution '0x0@0', codec '', calculated weight '0', valid url '1', has audio 'False', valid video 'windowed 0 fu
2021-08-19 09:24:17.478170 - StreamInfo - INFO --> llscreen 0', force UDP 'False'
2021-08-19 09:24:17.478391 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://192.168.1.11:10554/udp/av0_1', '-rtsp_transport', 'tcp']
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://192.168.1.11:10554/udp/av0_1', '-rtsp_transport', 'udp']
2021-08-19 09:24:18.346072 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://19
2021-08-19 09:24:18.346072 - StreamInfo - ERROR --> 2.168.1.11:10554/udp/av0_1', '-rtsp_transport', 'udp']' returned non-zero exit status 1.
2021-08-19 09:24:18.346752 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/udp/av0_1', resolution '0x0@0', codec '', calculated weight '0', valid url '1', has audio 'False', valid video 'windowed 0 fu
2021-08-19 09:24:18.346752 - StreamInfo - INFO --> llscreen 0', force UDP 'False'
2021-08-19 09:24:18.346934 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://192.168.1.13:10554/udp/av0_0', '-rtsp_transport', 'tcp']
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://192.168.1.13:10554/udp/av0_0', '-rtsp_transport', 'udp']
2021-08-19 09:24:19.215519 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '--', 'rtsp://19
2021-08-19 09:24:19.215519 - StreamInfo - ERROR --> 2.168.1.13:10554/udp/av0_0', '-rtsp_transport', 'udp']' returned non-zero exit status 1.
[... cut ...]
2021-08-19 09:24:36.221147 - ScreenManager - DEBUG - player watchdog
2021-08-19 09:24:51.359141 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-08-19 09:24:51.359567 - ScreenManager - DEBUG - player watchdog
2021-08-19 09:25:06.495009 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-08-19 09:25:06.495442 - ScreenManager - DEBUG - player watchdog
^C2021-08-19 09:25:16.062481 - Main - INFO - Exiting raspberry pi camplayer, have a nice day!

```

cybermaus, 2021-08-19:
So our replies crossed.
Maybe omxplayer eventually *also* has a problem.

But you told me the stream does not even show up in cache/streaminfo

So that to me is a clear indicator it is not even reaching the omxplayer part.
And I I have a few camera's myself where I cannot get omxplayer to work, but camplayer does make it work, by starting omxplayer witht he exact correct parameters. Which it figures out with the ffprobe.

cybermaus, 2021-08-19:
In your log, note that you see both tcp and udp ffprobe attempts printed, but never the result.
That means both calls fail.

However, I may see why now: the -rtsp-transport is placed *behind* the URL.

That is absolutely not allowed when you add the `--` like I made you do, but it is also not desirable normally.
Let me think a bit on how to fix that.

cybermaus, 2021-08-19:
I think like this:
```
                try:
                    ffprobe_args = ['ffprobe', '-v', 'error', '-show_entries',
                                    'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate']

                    if self.url.startswith('rtsp://'):
                        ffprobe_args.extend(['-rtsp_transport', transport])

                    ffprobe_args.extend(['--', self.url])
```

kareemtawab, 2021-08-19:
> So our replies crossed.
> Maybe omxplayer eventually _also_ has a problem.
>
> But you told me the stream does not even show up in cache/streaminfo
>
> So that to me is a clear indicator it is not even [...]

That was certainly unexpected 😁. My mistake eventually, I will try not forget to _quote reply_ the next time

And yes, cache/streaminfo is not present after running, even after adding `'--'`

Possible scenarios:
1. ffprobe does not deliver the arguments correctly.
2. omxplayer may have an issue with stream, but it is just me who is typing the command. Could be me too.

kareemtawab, 2021-08-19:
> I think like this:
>
> ```
>                 try:
>                     ffprobe_args = ['ffprobe', '-v', 'error', '-show_entries', [...]

I also added both prints.

```
2021-08-19 10:09:50.056414 - Main - INFO - Starting camplayer version 1.0.0b5
2021-08-19 10:09:50.056638 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-08-19 10:09:51.839429 - Main - INFO - ********************** SYSTEM INFO **********************
2021-08-19 10:09:51.839771 - Main - INFO - Camplayer version             = 1.0.0b5
2021-08-19 10:09:51.839897 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
2021-08-19 10:09:51.840050 - Main - INFO - Raspberry Pi SoC              = BCM2835
2021-08-19 10:09:51.840166 - Main - INFO - Raspberry Pi revision         = a02082
2021-08-19 10:09:51.840285 - Main - INFO - Raspberry Pi model name       = 3B
2021-08-19 10:09:51.840407 - Main - INFO - GPU memory allocation         = 256 MB
2021-08-19 10:09:51.840542 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-08-19 10:09:51.840672 - Main - INFO - VLC installed                 = True
2021-08-19 10:09:51.840793 - Main - INFO - pipng installed               = True
2021-08-19 10:09:51.840909 - Main - INFO - ffmpeg installed              = True
2021-08-19 10:09:51.841017 - Main - INFO - *********************************************************
2021-08-19 10:09:51.873462 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2021-08-19 10:09:51.873872 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2021-08-19 10:09:51.882237 - Window - DEBUG - init window with position '0 0 960 540', gridindex '[0, 1, 4, 5]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01'
2021-08-19 10:09:51.882743 - Window - DEBUG - init window with position '960 0 1920 540', gridindex '[2, 3, 6, 7]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W02'
2021-08-19 10:09:51.883046 - Window - DEBUG - init window with position '0 540 960 1080', gridindex '[8, 9, 12, 13]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03'
2021-08-19 10:09:51.883319 - Window - DEBUG - init window with position '960 540 1920 1080', gridindex '[10, 11, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W04'
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'tcp', '--', 'rtsp://192.168.1.11:10554/tcp/av0_0']
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'udp', '--', 'rtsp://192.168.1.11:10554/tcp/av0_0']
2021-08-19 10:10:12.022629 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport
2021-08-19 10:10:12.022629 - StreamInfo - ERROR --> ', 'udp', '--', 'rtsp://192.168.1.11:10554/tcp/av0_0']' timed out after 10 seconds
2021-08-19 10:10:12.023430 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/tcp/av0_0', resolution '0x0@0', codec '', calculated weight '0', valid url '1', has audio 'False', valid video 'windowed 0 fu
2021-08-19 10:10:12.023430 - StreamInfo - INFO --> llscreen 0', force UDP 'False'
2021-08-19 10:10:12.023608 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'tcp', '--', 'rtsp://192.168.1.11:10554/tcp/av0_1']
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'udp', '--', 'rtsp://192.168.1.11:10554/tcp/av0_1']
2021-08-19 10:10:32.094620 - StreamInfo - ERROR - ffprobe exception: Command '['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport
2021-08-19 10:10:32.094620 - StreamInfo - ERROR --> ', 'udp', '--', 'rtsp://192.168.1.11:10554/tcp/av0_1']' timed out after 10 seconds
2021-08-19 10:10:32.095821 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/tcp/av0_1', resolution '0x0@0', codec '', calculated weight '0', valid url '1', has audio 'False', valid video 'windowed 0 fu
2021-08-19 10:10:32.095821 - StreamInfo - INFO --> llscreen 0', force UDP 'False'
2021-08-19 10:10:32.096128 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'tcp', '--', 'rtsp://192.168.1.13:10554/tcp/av0_0']
['', '\ncodec_name=h264\ncodec_type=video\nwidth=1280\nheight=720\navg_frame_rate=25/1\nbit_rate=N/A\nmax_bit_rate=N/A\n[/STREAM]\n', '\ncodec_name=pcm_alaw\ncodec_type=audio\navg_frame_rate=0/0\nbit_rate=64000\nmax_bit_rate=N/A\n[/STREAM]\n']
2021-08-19 10:10:40.942664 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.13:10554/tcp/av0_0', resolution '1280x720@25', codec 'h264', calculated weight '23040000', valid url '1', has audio 'True', valid vide
[... cut ...]
2021-08-19 10:11:05.761785 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03' is not responding correctly, retrying within 250ms
2021-08-19 10:11:07.034921 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03' is not responding correctly, retrying within 250ms
2021-08-19 10:11:08.317679 - Window - WARNING - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03' is not responding correctly, retrying within 250ms
2021-08-19 10:11:09.591817 - Window - ERROR - DBus 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03' is not responding correctly after '6' attemps, give up now

```

cybermaus, 2021-08-19:
So the tcp one is not showing any result, and then the udp one is timing out.
The later is only to be expected. But the 20 second timing suggests they *both* time out after 10 seconds.

How long do your manual ffprobe attempts take? And your VLC connection time?

Changing timeout to 20 and also adding a print

```
                try:
                    ffprobe_args = ['ffprobe', '-v', 'error', '-show_entries',
                                    'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate']

                    if self.url.startswith('rtsp://'):
                        ffprobe_args.extend(['-rtsp_transport', transport])

                    ffprobe_args.extend(['--', self.url])

                    print(str(ffprobe_args))

                    # Invoke ffprobe, 20s timeout required for pi zero
                    streams = subprocess.check_output(ffprobe_args, universal_newlines=True, timeout=20,
                                                      stderr=subprocess.STDOUT).split("[STREAM]")

                    print('we definitely did reach post subprocess stream!')
                    print(str(streams))

```

kareemtawab, 2021-08-19:
> So the tcp one is not showing any result, and then the udp one is timing out.
> The later is only to be expected. But the 20 second timing suggests they _both_ time out after 10 seconds.
>
> How long do [...]

Takes around 5 seconds to get the stream going on VLC. Increased timeout to 20:

```
2021-08-19 10:53:37.841988 - Main - INFO - Starting camplayer version 1.0.0b5
2021-08-19 10:53:37.842249 - Main - INFO - Using config file '/home/pi/.camplayer/config.ini' and cache directory '/home/pi/.camplayer/cache/'
2021-08-19 10:53:39.562521 - Main - INFO - ********************** SYSTEM INFO **********************
2021-08-19 10:53:39.562814 - Main - INFO - Camplayer version             = 1.0.0b5
2021-08-19 10:53:39.562901 - Main - INFO - Operating system              = Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
2021-08-19 10:53:39.563011 - Main - INFO - Raspberry Pi SoC              = BCM2835
2021-08-19 10:53:39.563085 - Main - INFO - Raspberry Pi revision         = a02082
2021-08-19 10:53:39.563210 - Main - INFO - Raspberry Pi model name       = 3B
2021-08-19 10:53:39.563331 - Main - INFO - GPU memory allocation         = 256 MB
2021-08-19 10:53:39.563462 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2021-08-19 10:53:39.563591 - Main - INFO - VLC installed                 = True
2021-08-19 10:53:39.563701 - Main - INFO - pipng installed               = True
2021-08-19 10:53:39.563832 - Main - INFO - ffmpeg installed              = True
2021-08-19 10:53:39.563950 - Main - INFO - *********************************************************
2021-08-19 10:53:39.588495 - Main - INFO - Detected screen resolution for HDMI0 is '0x0@0Hz'
2021-08-19 10:53:39.588842 - Main - INFO - Using a virtual screen resolution of '1920x1080'
2021-08-19 10:53:39.595435 - Window - DEBUG - init window with position '0 0 960 540', gridindex '[0, 1, 4, 5]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W01'
2021-08-19 10:53:39.595871 - Window - DEBUG - init window with position '960 0 1920 540', gridindex '[2, 3, 6, 7]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W02'
2021-08-19 10:53:39.596136 - Window - DEBUG - init window with position '0 540 960 1080', gridindex '[8, 9, 12, 13]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W03'
2021-08-19 10:53:39.596369 - Window - DEBUG - init window with position '960 540 1920 1080', gridindex '[10, 11, 14, 15]', omxplayer dbus name 'org.mpris.MediaPlayer2.omxplayer_D01_S01_W04'
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'tcp', '--', 'rtsp://192.168.1.11:10554/udp/av0_1']
we definitely did reach post subprocess stream!
['', '\ncodec_name=h264\ncodec_type=video\nwidth=0\nheight=0\navg_frame_rate=0/0\nbit_rate=N/A\nmax_bit_rate=N/A\n[/STREAM]\n', '\ncodec_name=pcm_alaw\ncodec_type=audio\navg_frame_rate=0/0\nbit_rate=64000\nmax_bit_rate=N/A\n[/STREAM]\n']
2021-08-19 10:53:55.767445 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/udp/av0_1', resolution '0x0@0', codec 'h264', calculated weight '0', valid url '1', has audio 'True', valid video 'windowed 1
2021-08-19 10:53:55.767445 - StreamInfo - INFO -->  fullscreen 1', force UDP 'False'
2021-08-19 10:53:55.767942 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2021-08-19 10:53:55.771790 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.11:10554/udp/av0_1', resolution '0x0@0', codec 'h264', calculated weight '0', valid url '1', has audio 'True', valid video 'windowed 1
2021-08-19 10:53:55.771790 - StreamInfo - INFO -->  fullscreen 1', force UDP 'False'
2021-08-19 10:53:55.772130 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
['ffprobe', '-v', 'error', '-show_entries', 'stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate', '-rtsp_transport', 'tcp', '--', 'rtsp://192.168.1.13:10554/udp/av0_1']
we definitely did reach post subprocess stream!
['', '\ncodec_name=h264\ncodec_type=video\nwidth=0\nheight=0\navg_frame_rate=0/0\nbit_rate=N/A\nmax_bit_rate=N/A\n[/STREAM]\n', '\ncodec_name=pcm_alaw\ncodec_type=audio\navg_frame_rate=0/0\nbit_rate=64000\nmax_bit_rate=N/A\n[/STREAM]\n']
2021-08-19 10:54:12.087188 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.13:10554/udp/av0_1', resolution '0x0@0', codec 'h264', calculated weight '0', valid url '1', has audio 'True', valid video 'windowed 1
2021-08-19 10:54:12.087188 - StreamInfo - INFO -->  fullscreen 1', force UDP 'False'
2021-08-19 10:54:12.087637 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2021-08-19 10:54:12.091773 - StreamInfo - INFO - stream properties 'rtsp://192.168.1.13:10554/udp/av0_1', resolution '0x0@0', codec 'h264', calculated weight '0', valid url '1', has audio 'True', valid video 'windowed 1
2021-08-19 10:54:12.091773 - StreamInfo - INFO -->  fullscreen 1', force UDP 'False'
2021-08-19 10:54:12.092167 - StreamInfo - INFO - RUN 'camplayer --rebuild-cache' IF THIS STREAM INFORMATION IS OUT OF DATE!!
2021-08-19 10:54:12.093281 - Screen - INFO - init screen number '1' with a total weight of '0'
2021-08-19 10:54:12.093540 - ScreenManager - INFO - added screen number '1' to display '1' with layout '4' and displaytime '10'
[... cut ...]
2021-08-19 10:54:14.069375 - ScreenManager - DEBUG - stream/player health checking for display number '1'
2021-08-19 10:54:14.069657 - ScreenManager - DEBUG - player watchdog
```

and cache

```
{
    "rtsp://192.168.1.13:10554/udp/av0_1": {
        "codec_name": "h264",
        "height": 0,
        "width": 0,
        "framerate": 0,
        "audio": true,
        "force_udp": false
    },
    "rtsp://192.168.1.11:10554/udp/av0_1": {
        "codec_name": "h264",
        "height": 0,
        "width": 0,
        "framerate": 0,
        "audio": true,
        "force_udp": false
    }
}
```

cybermaus, 2021-08-19:
Ok, so the tcp stream was indeed timing out.
Now, it no longer tries the udp stream. And it prints a result. And it builds the cache.

Granted, the cache is mostly 0, and while you do not mention, yet I am assuming it is not actually playing.
But the the tcp definitely timed out, it needed a longer time.

And still, the manual ffprobe did work, by your suggestion:

> They all output:
>
>   Metadata:
>     title           : streamed by the Zmodo RTSP server
>   Duration: N/A, start: 0.040000, bitrate: N/A
>     Stream #0:0: Video: h264 (Baseline), yuv420p(progressive), [...]

So lets try a bit more with the exact format.
Also, like I suggested before, can we start using the main stream av0_0

You already tried, but please try again:

`ffprobe -v error -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -rtsp_transport tcp -- rtsp://192.168.1.11:10554/tcp/av0_0`

Oh, and since your cache/streaminfo is now populated, but with incorrect data, you will have to delete it on each attemp.

kareemtawab, 2021-08-19:
YAAAY!! I actually got omxplayer to work manually using:

`omxplayer --aidx=-1 --avdict=rtsp_transport:udp  --win '0 0 800 480' --live -- rtsp://192.168.1.13:10554/udp/av0_1`

I had to disable the audio stream by `--aidx=-1`, change the syntax from `rtsp_transport udp` to `--avdict=rtsp_transport:udp`, added the dimensions of the window `--win '0 0 800 480'` and at last added the `--live` for un-seekable streams.

Not sure if I will be able to implement `--aidx=-1`,  `--avdict=rtsp_transport:udp` and `--live` in camplayer, though. Let me try that now.

kareemtawab, 2021-08-19:
> Ok, so the tcp stream was indeed timing out.
> Now, it no longer tries the udp stream. And it prints a result. And it builds the cache.
>
> Granted, the cache is mostly 0, and while you do not mention, [...]

After I got omxplayer using the av0_1, I think it doesn't help to use the av0_0 as it causes broken stream output in omxplayer because of limited bandwidth (as I saw from the output) but I will add it anyway to the config.

`ffprobe -v error -show_entries stream=codec_type,height,width,codec_name,bit_rate,max_bit_rate,avg_frame_rate -rtsp_transport tcp -- rtsp://192.168.1.11:10554/tcp/av0_0`

gives:

```
[STREAM]
codec_name=h264
codec_type=video
width=1280
height=720
avg_frame_rate=25/1
bit_rate=N/A
max_bit_rate=N/A
[/STREAM]
[STREAM]
codec_name=pcm_alaw
codec_type=audio
avg_frame_rate=0/0
bit_rate=64000
max_bit_rate=N/A
[/STREAM]

```

cybermaus, 2021-08-19:
Ok. so why does it give proper output when run manually, and almost all 0 when running via camplayer?

I do have a workaround suggestion for you though:
Fake the cache/streaminfo with the proper values. Once the cache/streaminfo exists, camplayer does not even try the ffprobe anymore, and goes onward to omxplayer. And since it now did work for omxplayer....

kareemtawab, 2021-08-19:
> Ok. so why does it give proper output when run manually, and almost all 0 when running via camplayer?
>
> I do have a workaround suggestion for you though:
> Fake the cache/streaminfo with the proper [...]

Could be possible that the arguments sent by ffprobe is not what omxplayer expects?

cybermaus, 2021-08-20:
So any luck with faking the cache/streaminfo with the correct values, so it skips ffprobe and goes directly to camplayer?

middletonzx1fn0, 2021-10-05:
I also had the same issue with 10 secs drop signal. Has this been fix or is there plan on updating the repo to solve the omxplayer udp issue?