raspicamplayer/camplayer #65: Configuring Two screen layouts in one screnen

SistaSter, 2021-08-20:
Would it be possible to configure two screen layaouts to view only two cameras? 
I have configured with 4 layeouts , I Need only see this two camaras.

[image]
Thanks for all.

cybermaus, 2021-08-20:
You can probably trick it by specifying a virtual resolution in .camplayer/config.ini that is either twice as wide, or twice as high as your actual screen, depending if you want the displays above each other, or to the side of each other.

Myself, I have made a [modified version](https://github.com/cybermaus/camplayer) for myself that supports 1x2 and 1x3 layouts. However, to get a somewhat reasonable view, you need to rotate your screen and the raspberry pi display 90 degrees (or 270 degrees). This is also visible in [this guys change request](https://github.com/raspicamplayer/camplayer/pull/13) (not me, but similar). 

Do note that when most LCD screen viewing angles are optimised to be viewed from left to right, and somewhat from above, but not at all from below. So when you physically rotate the screen by 90 or 270 degrees, it is usually less visible from one side, and even worse from the other side.

cybermaus, 2021-08-20:
PS: that right picture looks suspiciously like my fathers potato sorting machine, taking me back 40+ years....

dwasyl, 2021-12-03:
@cybermaus For your rotated displays, do you think you could have just stretched the image rectangular? For example, I'm trying to get two displays to display in a "long" format (like 600x800 type thing). One displays fine, but two is an issue.

cybermaus, 2022-01-10:
Sorry, late reaction. Yes, you can just stretch it rectangular, but you would still either need an mod to have a 2 screen division, or have a "fake screen" of 4 with 2 of the screens stretched out of the actual screen.

dwasyl, 2022-01-18:
All good! How do I get it to stretch rectangular that way? I couldn't find the right setting. Probably easier to have the 4 screens with 2 fake ones rather than a two screen division.