raspicamplayer/camplayer #69: Broken Pipe

yonasRP, 2021-09-01:
When trying to run the demo using:

camplayer --demo

i get the following error:

2021-09-01 10:07:28.026949 - BackGroundManager - DEBUG - setting background '/home/pi/.camplayer/cache/nolink_1P5_1920_1080.png' visible for display '0
Traceback (most recent call last):
  File "./camplayer.py", line 313, in <module>
    main()
  File "./camplayer.py", line 232, in main
    screenmanager.do_work()
  File "/usr/local/share/camplayer/camplayer/screenmanager.py", line 1183, in do_work
    self._screen_rotate_next_active(display_idx=display_idx)
  File "/usr/local/share/camplayer/camplayer/screenmanager.py", line 992, in _screen_rotate_next_active
    self._screens[display_idx][self._active_screen_idx[display_idx]].layout), display_idx=display_idx)
  File "/usr/local/share/camplayer/camplayer/backgroundgen.py", line 286, in show_background
    cls._proc_background[display_idx].stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe

Any help is appreciated.

cybermaus, 2021-09-01:
Many things can lead to "broken pipes" but for me (with the same "in show_background" context) it is because the method of displaying background and icons is somewhat instable.

Solved definitely by turning them off:

```
[ADVANCED]
backgroundmode = 3
icons = 0
```

But I also had some success by turning off only one, or by setting background to 1
```
[ADVANCED]
backgroundmode = 1
icons = 1
```

ulrichanthony, 2021-09-30:
You need to add these to the config.ini file.