raspicamplayer/camplayer #93: This file is currently set to private - https://www.rpi-camplayer.com/download/

rtek1000, 2022-01-17:
Hello,

Camplayer OS (the plug and play solution)

Camplayer OS is a plug and play operating system containing the camplayer software and more.
Its main goal is to improve the reliability and user experience.

Camplayer OS is currently in beta phase.
Click here to download <---- **This file is currently set to private**

https://www.rpi-camplayer.com/download/

HerdOn304, 2022-02-10:
I have this same issue. Can we get this addressed please

dkramarc, 2022-03-23:
Even if you donate you don't have access. Sent e-mail a few times regarding access and just get ignored. )-:
Shame as it's a great product (and why I donated).

HerdOn304, 2022-03-23:
@dkramarc I also donated and still didn't get access