raspicamplayer/camplayer #97: Unsupported hardware with revision b03115

theantnest, 2022-03-03:
The new raspberry Pi 4B v1.5 is throwing an unsupported hardware error when trying to run the test/ demo.

theantnest, 2022-03-03:
`hardwarecheck = 0`

does not solve it

paciops, 2022-03-08:
You can [edit utils.py](https://github.com/paciops/camplayer/commit/21aaa23059693bba292a8678f9aa00fbc8b2a6d3) as I have done (if you don't want to wait) to make it work

theantnest, 2022-03-08:
I tried this, but must be doing something wrong. I forked the project, edited the file, then pulled the install files from my forked repo but it didn't work.

Is there a better way?

theantnest, 2022-03-08:
Not sure if the owner of this project has abandoned it? Hope he or she is all good.

paciops, 2022-03-08:
> I tried this, but must be doing something wrong. I forked the project, edited the file, then pulled the install files from my forked repo but it didn't work.
> 
> Is there a better way?

It should work,  can you post the logs?

theantnest, 2022-03-10:
Thanks for trying to help me out. I have no idea what I was doing wrong, but now I have it sorted and running!

TheSpawnMan, 2022-08-31:
i installed from your repository with the fix applied and that didnt work same error

paciops, 2022-08-31:
@TheSpawnMan mine or @theantnest  version? And which error?

theantnest, 2022-08-31:
> i installed from your repository with the fix applied and that didnt work same error

This has been working rock solid for me running 24/7 for about 4 months straight. Haven't even restarted a pi!

TheSpawnMan, 2022-08-31:
```
camplayer --demo
2022-08-31 11:37:23.161181 - Main - INFO - Starting camplayer version 1.0.0b5
2022-08-31 11:37:23.161331 - Main - INFO - Using config file '../examples/demo-config.ini' and cache directory '/home/nspohn/.camplayer/cache/'
2022-08-31 11:37:24.899036 - Main - INFO - ********************** SYSTEM INFO **********************
2022-08-31 11:37:24.899575 - Main - INFO - Camplayer version             = 1.0.0b5
2022-08-31 11:37:24.899804 - Main - INFO - Operating system              = Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
2022-08-31 11:37:24.900125 - Main - INFO - Raspberry Pi SoC              = BCM2711
2022-08-31 11:37:24.900350 - Main - INFO - Raspberry Pi revision         = b03115
2022-08-31 11:37:24.900552 - Main - INFO - Raspberry Pi model name       =
2022-08-31 11:37:24.900761 - Main - INFO - GPU memory allocation         = 256 MB
2022-08-31 11:37:24.900982 - Main - INFO - Python version                = 3.7.3 (default, Jan 22 2021, 20:04:44)  MB
2022-08-31 11:37:24.901205 - Main - INFO - VLC installed                 = False
2022-08-31 11:37:24.901404 - Main - INFO - pipng installed               = True
2022-08-31 11:37:24.901602 - Main - INFO - ffmpeg installed              = True
2022-08-31 11:37:24.901787 - Main - INFO - *********************************************************
Unsupported hardware with revision b03115 ...
```

TheSpawnMan, 2022-08-31:
> ```
> camplayer --demo
> 2022-08-31 11:37:23.161181 - Main - INFO - Starting camplayer version 1.0.0b5
> 2022-08-31 11:37:23.161331 - Main - INFO - Using config file '../examples/demo-config.ini' and [...]

> @TheSpawnMan mine or @theantnest version? And which error?
i pulled from @paciops

paciops, 2022-09-01:
okay try to post also the `camplayer/utils/utils.py`

TheSpawnMan, 2022-09-01:
id love to.... how do i find where its installed at to access that file or modify it

paciops, 2022-09-03:
depends where you cloned the repo the `camplayer` folder is the folder of this project

smwhtdmgd13, 2022-10-10:
Sorry to rehash this but is it possible to just modify the camplayer/utils/utils.py file instead of pulling from another already modified repository?  I found the file, added another line for b03115, rebooted, and it didn’t change the error so I’m guessing not?

daveparisee, 2023-08-03:
Found it..
Edit /usr/local/share/camplayer/camplayer/utils/utils.py instead of installation files.... :| 
Work A1 now