raspicamplayer/camplayer #108: Fixed Building

RamboRogers, 2024-05-26:
I forked this and updated the readme so you can build and it works for anyone looking here.

https://github.com/RamboRogers/camplayer

jesjhoward, 2024-06-19:
Can't seem to get your version working. I'm not 100% sure, but it seems to be a problem with piping? --demo never displays to the screen and ends with:
Traceback (most recent call last):
  File "./camplayer.py", line 313, in <module>
    main()
  File "./camplayer.py", line 232, in main
    screenmanager.do_work()
  File "/usr/local/share/camplayer/camplayer/screenmanager.py", line 1183, in do_work
    self._screen_rotate_next_active(display_idx=display_idx)
  File "/usr/local/share/camplayer/camplayer/screenmanager.py", line 992, in _screen_rotate_next_active
    self._screens[display_idx][self._active_screen_idx[display_idx]].layout), display_idx=display_idx)
  File "/usr/local/share/camplayer/camplayer/backgroundgen.py", line 286, in show_background
    cls._proc_background[display_idx].stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe

Any help?

RamboRogers, 2024-06-19:
did you follow the instructions and run that old buster release?

2wenty2wo, 2024-11-02:
Worked perfectly, thank you.