raspicamplayer/camplayer #113: Appoint successor for this project - 2025

bitvilag, 2025-06-24:
Dear All,

I would like to officially choose a successor of this project.

Its clearly seen that this project is abandoned, but there are still a lot to do with the project.

When choosing I would like to put in my fork a an option for a possible landingZone for this project. Did not yet do too many changes.
Also I am open for suggestions for other forks.

My guess the next big thing is moving to vlc or cvlc?

bitvilag, 2025-07-28:
> or you can switch to https://github.com/OpenSurv/OpenSurv though it can run on all sort of hardware, not optimized for raspberry pi

I have tested it and so far I found one issue which I mentioned in a [discussion ](https://github.com/OpenSurv/OpenSurv/discussions/61)as well. The issue is related to the window title bar where the minimize, maximize, title is as well. I cannot seem to remove it. Messes up the whole layout.

tim661811, 2025-09-30:
I used this project a few years ago and was disappointed to see it no longer works. After trying to get it running using the legacy OMXPlayer archives and experimenting with OpenSurv — both unsuccessfully — I decided to create my own simplified version based on VLC. This version does **not support multiple streams on one screen**, but it might still be useful for others. I’ve uploaded it to GitHub: [raspberry-pi-rtsp-monitor](https://github.com/tim661811/raspberry-pi-rtsp-monitor).

cybermaus, 2025-09-30:
> I used this project a few years ago and was disappointed to see it no longer works. After trying to get it running using the legacy OMXPlayer archives and experimenting with OpenSurv — both [...]

Nice. Alas, I need split screen.
Mind you, still using camplayer.  Got 3 monitors with 4-way split screen each in various locations around my brothers campground. The trick: Don't update anything, especially not RasPi, run a 6yo copy. They are on a isolated network anyway. Clone the card if you want to rebuild one.